mne-tools/mne-python #6338: [RFC] Draft for the new release note v0.18

GuillaumeFavelier, 2019-05-18:
I put together a release note, please feel free to comment on it (or edit).

--- 

Dear community,

We are very pleased to announce the new 0.18 release of MNE-Python (http://martinos.org/mne/stable/).

A few highlights
================

* to be added

Notable API changes
===================

* to be added

For a full list of improvements and API changes, see:
http://martinos.org/mne/stable/whats_new.html#version-0-18

To install the latest release the following command should do the job:

    $ pip install --upgrade mne

As usual, we welcome your bug reports, feature requests, critiques, and contributions.

Some links:
- https://github.com/mne-tools/mne-python (code + readme on how to install)
- http://martinos.org/mne/stable/ (full MNE documentation)

Follow us on Twitter: https://twitter.com/mne_news

Regards,
The MNE-Python developers

52 people made commits that contributed to this release (in alphabetical order):

* 	Achilleas Koutsou
* 	Alexander Kovrig
* 	Alexandre Gramfort
* 	Antoine Gauthier
* 	Britta Westner
* 	Bruno Nicenboim
* 	Burkhard Maess
* 	Chris Bailey
* 	Chris Holdgraf
* 	Christian Brodbeck
* 	Clemens Brunner
* 	Cristóbal Moënne-Loccoz
* 	Daniel McCloy
* 	David Haslacher
* 	Denis A. Engemann
* 	Dirk Gütlin
* 	Eric Larson
* 	Evgenii Kalenkovich
* 	Fede Raimondo
* 	Guillaume Favelier
* 	Hubert Banville
* 	Ivana Kojcic
* 	Jean-Remi King
* 	Jeff Hanna
* 	Joan Massich
* 	Jon Houck
* 	Jona Sassenhagen
* 	José C. García Alanis
* 	Jussi Nurminen
* 	Katarina Slama
* 	Keith Doelling
* 	Larry Eisenman
* 	Legrand Nicolas
* 	Mainak Jas
* 	Marijn van Vliet
* 	Mikolaj Magnuski
* 	Nathalie Gayraud
* 	Nikolas Chalas
* 	Oleh Kozynets
* 	Quentin Bertrand
* 	Samuel Deslauriers-Gauthier
* 	Sebastián Castaño
* 	Simon Kern
* 	Stanislas Chambon
* 	Stefan Appelhoff
* 	Stefan Repplinger
* 	Steve Matindi
* 	Teon Brooks
* 	Thomas Donoghue
* 	Thomas Hartmann
* 	jeythekey
* 	kostiantyn maksymenko

sappelhoff, 2019-05-18:
We could add the second twitter channel:

```Text
Follow us on Twitter for general news: https://twitter.com/mne_news
... and for for a regular overview of merged PRs: https://twitter.com/mne_python

```

britta-wstnr, 2019-05-18:
Something like:
``Better support for source reconstruction with beamformers // the beamformer module and other inverse models`` ?
-- beamformers are inverse models, too, that does not get quite clear to me as it is now.

massich, 2019-05-18:
is it better now?

sappelhoff, 2019-05-18:
And we catched and fixed more than 50 bugs !! --> `caught`

britta-wstnr, 2019-05-18:
@massich sounds better to me, thanks!

britta-wstnr, 2019-05-18:
``Add*ed* overwrite parameter`` under API changes? The rest is all past tense.

jasmainak, 2019-05-19:
> catched and fixed more than 50 bugs

a naive reader might assume that it means mne-python is full of bugs :p Maybe you want to phrase it a little differently?

agramfort, 2019-05-19:
made a few edits. LGTM

cbrnr, 2019-05-20:
I also made some minor edits (formatting). LGTM!

larsoner, 2019-05-20:
We should probably cut a 0.18.1 with the `mne.realtime` deprecations included, and add it to the release notes here

massich, 2019-05-20:
I propose to do bump to `0.18.1rc0` in `maint/0.18` and do a pre-release waiting to get in some earlier feedback.

I tried in `test.pypi.org` but I don't seem to be able to find my package.
```sh
~/code/mne-python maint/0.18*
(mne) ❯ python setup.py sdist                                     
running sdist
running egg_info
writing mne.egg-info/PKG-INFO
writing dependency_links to mne.egg-info/dependency_links.txt
writing requirements to mne.egg-info/requires.txt
writing top-level names to mne.egg-info/top_level.txt
reading manifest file 'mne.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'mne/datasets' under directory 'mne'
warning: no previously-included files matching '*' found under directory 'examples/MNE-sample-data'
warning: no previously-included files matching '*' found under directory 'examples/MNE-testing-data'
warning: no previously-included files matching '*' found under directory 'examples/MNE-spm-face'
warning: no previously-included files matching '*' found under directory 'examples/MNE-somato-data'
warning: no previously-included files found matching 'Makefile'
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '*.yml'
warning: no previously-included files found matching 'ignore_words.txt'
warning: no previously-included files found matching '.mailmap'
warning: no previously-included files matching '*' found under directory 'doc'
warning: no previously-included files matching '*' found under directory 'logo'
warning: no previously-included files found matching 'CODE_OF_CONDUCT.md'
warning: no previously-included files found matching '.github'
warning: no previously-included files found matching '.github/CONTRIBUTING.md'
warning: no previously-included files found matching '.github/ISSUE_TEMPLATE'
warning: no previously-included files found matching '.github/ISSUE_TEMPLATE/blank.md'
warning: no previously-included files found matching '.github/ISSUE_TEMPLATE/bug_report.md'
warning: no previously-included files found matching '.github/ISSUE_TEMPLATE/feature_request.md'
warning: no previously-included files found matching '.github/PULL_REQUEST_TEMPLATE.md'
warning: no previously-included files matching '*' found under directory 'mne/io/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/io/bti/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/io/edf/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/io/kit/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/io/brainvision/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/io/egi/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/io/nicolet/tests/data'
warning: no previously-included files matching '*' found under directory 'mne/preprocessing/tests/data'
writing manifest file 'mne.egg-info/SOURCES.txt'
running check
creating mne-0.18.1rc0
creating mne-0.18.1rc0/bin
creating mne-0.18.1rc0/examples
creating mne-0.18.1rc0/examples/connectivity
creating mne-0.18.1rc0/examples/datasets

       ..... TL;DR;  ...

Writing mne-0.18.1rc0/setup.cfg
Creating tar archive
removing 'mne-0.18.1rc0' (and everything under it)

~/code/mne-python maint/0.18* 32s
(mne) ❯ twine upload --repository-url https://test.pypi.org/legacy/ dist/mne-0.18.1rc0.tar.gz 
Enter your username: massich
Enter your password: 
Uploading distributions to https://test.pypi.org/legacy/
Uploading mne-0.18.1rc0.tar.gz
100%|████████████████████████████████████████████████████████████████████████████████████████████████████████| 6.03M/6.03M [00:03<00:00, 1.73MB/s]

massich, 2019-05-20:
It does not allow me to push it back, so it means is there. But I can't find it in test.pypi.org

closed 2019-05-22 by massich