mne-tools/mne-python #6343: Can't save a slice of a bdf file without preloading full file

Pezz89, 2019-05-19:
Having updated MNE to v0.18.0 via pip, I'm now unable to split a .bdf file into smaller raw.fif files without preloading the full file. This code runs in MNE v0.17.2.
```
In [1]: import mne

In [2]: raw = mne.io.read_raw_edf('./path/to/eeg_file.bdf')
Extracting EDF parameters from ./path/to/eeg_file.bdf...
BDF file detected
Setting channel info structure...
Creating raw.info structure...

In [3]: raw.save('./test-raw.fif', tmin=1.2, tmax=4.0, overwrite=True)
Overwriting existing file.
Writing /home/sam/temp/test-raw.fif
Closing /home/sam/temp/test-raw.fif [done]
```
However, as of 0.18.0, this code returns the following error:
```
~/.pyenv/versions/3.6.8/lib/python3.6/site-packages/mne/io/edf/edf.py in _read_segment_file(data, idx, fi, start, stop, raw
_extras, chs, filenames)
    414
    415     if subtype == 'bdf':
--> 416         cal[0, stim_channel_idx] = 1
    417         offsets[stim_channel_idx, 0] = 0
    418         gains[0, stim_channel_idx] = 1
```
The error occurs as the `stim_channel_idx` variable is set to:
```
ipdb> stim_channel_idx
array([], dtype=float64)
```
which I believe occurs in the block of code on lines 407-413 of `/mne/io/edf/edf.py`. Unfortunately I don't have a good enough understanding of this code block to offer any more information. 

This issue does not occur if the file has been preloaded using `preload=True` in the call to `read_raw_edf`. Unfortunately this is not an option in some cases where the .bdf files are simply too large to load fully in to memory.

agramfort, 2019-05-19:
I cannot replicate the pb with
https://github.com/mne-tools/mne-testing-data/tree/master/BDF:

In [10]: raw = mne.io.read_raw_bdf('test_generator_2.bdf')

In [11]: raw.save('toto_raw.fif', tmin=1., tmax=2, overwrite=True)

can you share a file?

do I miss something?

>

Pezz89, 2019-05-19:
Unfortunately I cannot share the full files, but will try to generate a small test file that reproduces the issue in the next few days (I don't currently have access to software that can crop BDF data and will need to find a Windows or Matlab install to do this).

agramfort, 2019-05-20:
you can replicate the pb only with your file? not any of the files from our
testing repo?

>

Pezz89, 2019-08-26:
Sorry for the long delay on getting around to this, but would you be able to point me towards the test files? I'll give them a go and see if it's an issue that only affects my files.
Even if that is the case maybe it would useful just to have a more intuitive error message to suggest what the problem with the file is.

cbrnr, 2019-08-26:
The test files are at the URL @agramfort posted above.

Pezz89, 2019-08-28:
Sorry I missed that link. Ok so I've now run the exact same code using mne 18.2 straight off of pip, on a different completely clean brand new machine running Ubuntu 18.04.3 LTS. It appears that the bug occurs for one of the two test stimulus I was linked too ('test_bdf_stim_channel.bdf').

```
In [1]: import mne

In [2]: raw = mne.io.read_raw_edf('./test_bdf_stim_channel.bdf')

In [3]: raw                                  
Out[3]: <RawEDF  |  test_bdf_stim_channel.bdf, n_channels x n_times : 4 x 5000 (10.0 sec), ~12 kB, data not loaded>

In [4]: raw.save('./test-raw.fif', tmin=1.2, tmax=4.0, overwrite=True)
---------------------------------------------------------------------------                                                                                                                                                                                                                                       
IndexError                                Traceback (most recent call last)                                                                                                            
<ipython-input-3-cef6f7eef39b> in <module>                                                                                                                                                                                                                                                                        
----> 1 raw.save('./test-raw.fif', tmin=1.2, tmax=4.0, overwrite=True)                                                                                                                                                               
                                                                                           
</home/bp/.pyenv/versions/anaconda3-2019.03/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-141> in save(self, fname, picks, tmin, tmax, buffer_size_sec, drop_small_buffer, proj, fmt, overwrite, split_size, split_naming, verbose)
                                                                                                                                                                                                                                                                                                                  
~/.pyenv/versions/anaconda3-2019.03/lib/python3.7/site-packages/mne/utils/_logging.py in wrapper(*args, **kwargs)                                                                                                                                                                                                 
     87             with use_log_level(verbose_level):                                                                                                   
     88                 return function(*args, **kwargs)                                                                                                                                                                                                                                                          
---> 89         return function(*args, **kwargs)                                                                                                         
     90     return FunctionMaker.create(                                                                                                                 
     91         function, 'return decfunc(%(signature)s)',                                                                                               
                                                                                                                                                         
~/.pyenv/versions/anaconda3-2019.03/lib/python3.7/site-packages/mne/io/base.py in save(self, fname, picks, tmin, tmax, buffer_size_sec, drop_small_buffer, proj, fmt, overwrite, split_size, split_naming, verbose)                                                                                               
   1595         data_type = type_dict[fmt]                                  
   1596                                                                                                                                                  
-> 1597         data_test = self[0, 0][0]                                   
   1598         if fmt == 'short' and np.iscomplexobj(data_test):                                                                                        
   1599             raise ValueError('Complex data must be saved as "single" or '                                                                        

~/.pyenv/versions/anaconda3-2019.03/lib/python3.7/site-packages/mne/io/base.py in __getitem__(self, item)                                                                                                                                                                                                         
    915             data = self._read_segment(start=start, stop=stop, sel=sel,                                                                           
    916                                       projector=self._projector,                                                                                 
--> 917                                       verbose=self.verbose)                                                                                      
[... cut ...]
BDF file detected
Setting channel info structure...
Creating raw.info structure...

In [6]: raw.save('./test-raw.fif', tmin=1.2, tmax=4.0, overwrite=True)      
Writing /home/bp/Work/SOTON/PhD/SOTON_Experiment_1_analysis/mne_bug_test/test-raw.fif
Closing /home/bp/Work/SOTON/PhD/SOTON_Experiment_1_analysis/mne_bug_test/test-raw.fif [done]

In [7]: raw                           
Out[7]: <RawEDF  |  test_generator_2.bdf, n_channels x n_times : 11 x 120000 (600.0 sec), ~26 kB, data not loaded>

```

agramfort, 2019-08-29:
@cbrnr shall I let you have a look?

cbrnr, 2019-08-29:
Yes, I can take a look.

cbrnr, 2019-08-29:
The commit that broke it is here:
https://github.com/mne-tools/mne-python/commit/42cf6876414b9b3d1ad97882962110ae5cbd908d#diff-825d45683d0be02fad23614151867e02

This was part of #5841 - @massich can you take over please?

agramfort, 2019-08-29:
see https://github.com/mne-tools/mne-python/pull/6720

PR #6720 (2019-08-30): Fix bdf slice
closes #6343
closed 2019-08-30 by cbrnr