mne-tools/mne-python #6344: Bug: plot_compare_evokeds, issue with Neuromag-122-data (rename_channels)

marsipu, 2019-05-19:
#### Describe the bug
When I try to do the mne.viz.plot_compare_evokeds() with data from our Neuromag122-MEG (only planar-gradiometers), the following Error-Message occurs:
ValueError: New channel names are not unique, renaming failed

The traceback goes to rename_channels which is called within combine_grads.
I assume, the problem is, that the naming of the channels is here very simple:
MEG 001, MEG 002, MEG 003, [...] MEG 122

In _combine_grad(1545), the last letter of the ch_name ist changed to 'X', which leads for this data to the same ch_names.

I didn't understand the purpose of the 'X', so I hope, you can help me with fixing this.
Below is a code snippet with some Neuromag-122-data.
Thank you!

#### Steps and/or code to reproduce
[Neuromag122_test.zip](https://github.com/mne-tools/mne-python/files/3195652/Neuromag122_test.zip)

import mne
from os.path import join

path_to_file = '<your_path_to_test_file>'
data_path = join(path_to_file, 'Neuromag122_test-ave.fif')

evokeds = mne.read_evokeds(data_path)
mne.viz.plot_compare_evokeds(evokeds)

agramfort, 2019-05-20:
I confirm the pb. Here is the test to use for the PR:

```
import os.path as op
import mne
from mne.io.constants import FIFF

base_dir = op.join(op.dirname(mne.__file__), 'io', 'tests', 'data')
evoked_fname = op.join(base_dir, 'test-ave.fif')

def test_plot_compare_evokeds_neuromag122():
    """Test topomap plotting."""
    evoked = mne.read_evokeds(evoked_fname, 'Left Auditory',
                              baseline=(None, 0))
    evoked.pick_types(meg='grad')
    evoked.pick_channels(evoked.ch_names[:122])
    ch_names = ['MEG %03d' % k for k in range(1, 123)]
    for c in evoked.info['chs']:
        c['coil_type'] = FIFF.FIFFV_COIL_NM_122
    evoked.rename_channels({c_old: c_new for (c_old, c_new) in
                            zip(evoked.ch_names, ch_names)})
    mne.viz.plot_compare_evokeds([evoked, evoked])

test_plot_compare_evokeds_neuromag122()
```

jona-sassenhagen, 2019-05-28:
So the issue is about `_combine_grads`/`combine_evokeds`, correct? In which case I'd pass. If it's about `plot_compare_evokeds`, assign me ...

agramfort, 2019-06-09:
I looked at the issue and to me you should not have to combine_grads when you show line plots like `plot_compared_evokeds`. Looking in more details to the API logic the fact that the choice of picks overwrites the gfp parameter is nuts. Another weirdness is that `gfp=False` means average.... I don't know what's the best way forward but we should aim for:

```
mne.viz.plot_compare_evokeds([evoked1, evoked2], picks='meg')
mne.viz.plot_compare_evokeds([evoked1, evoked2], picks=None)
```

returning the same thing. And have something like:

```
mne.viz.plot_compare_evokeds([evoked1, evoked2], picks='meg', combine='mean')
mne.viz.plot_compare_evokeds([evoked1, evoked2], picks=None, combine='mean')
```

where combine follows the API of `plot_epochs_image` ?

any thoughts?

wdyt @jona-sassenhagen ?

jona-sassenhagen, 2019-06-09:
Yes, I think it's a good idea to follow @drammock 's work on `plot_epochs_image` here.

I will not have the time to do this any time soon myself I fear.

agramfort, 2019-06-09:
I did a minimal fix in https://github.com/mne-tools/mne-python/pull/6438

drammock, 2019-06-09:
I realized late on friday that my work on `plot_epochs_image` will likely need to heavily touch `plot_compare_evokeds` too (in part because of how merging grads is handled), so I will keep @agramfort's comments here in mind while working on #6422.

agramfort, 2019-06-10:
ok I am merging #6438. @drammock let's follow up on PR on plot_epochs_image

PR #6438 (2019-06-10): fix vectorview 122 with plot_compare_evoked
closes #6344
closed 2019-06-10 by agramfort