mne-tools/mne-python #6345: No stim channels found. Consider specifying them manually using the 'stim_channel' parameter.

h2oz, 2019-05-20:
Hi,
I got an error when I call 
_mne.find_events(raw=raw, shortest_event=1, verbose=False)_ that I didn't get previously:

_No stim channels found. Consider specifying them manually using the 'stim_channel' parameter._

Why is that? I went back to version 0.17.2

cbrnr, 2019-05-20:
What kind of file did you load? MNE changed the way it imports markers - most likely you can find them in `raw.annotations` now.

h2oz, 2019-05-20:
I use this database with gdf files: [Brain Invaders 2013a] [https://zenodo.org/record/1494240#.XOKC_6Q6-w5](url)

> Data were acquired with a Nexus (TMSi, The Netherlands) EEG amplifier:
>     Sampling frequency: 512 samples per second
>     Digital Filter: No
>     Electrodes:  16 wet Silver/Silver Chloride electrodes positioned at FP1, FP2, F5, AFz, F6, T7, Cz, T8, P7, P3, Pz, P4, P8, O1, Oz, O2 according to the 10/20 international system

cbrnr, 2019-05-20:
Yes, GDF events are now accessible via `raw.annotations`:

```
import mne
raw = mne.io.read_raw_gdf("1.gdf", preload=True)
```

If you inspect `raw.annotations`, you will find all events in there.

cbrnr, 2019-05-20:
If you want to have events, you can convert annotations to events with `mne.events_from_annotations(raw)` (there are several parameters that influence the mapping from annotation to event, so be sure to check them out).

h2oz, 2019-05-20:
Great thanks!

closed 2019-05-20 by h2oz

agramfort, 2019-05-20:
the mne.find_events function should have told you to use
events_from_annotations if you have no stim channel and annotations....

can someone open a PR?

>

PR #6347 (2019-05-21): Better find_events message if no stim channel is found
Fixes #6345. OK @agramfort?