mne-tools/mne-python #6361: [Feature request] Store trials marked as bad

Eklund89, 2019-05-22:
#### Context
To visually reject trials based on peak to peak values, I made a simplified version of ft_rejectvisual (http://www.fieldtriptoolbox.org/reference/ft_rejectvisual/) for MNE-python.
This function spits out a list of the trials i want to drop so i can drop them with epochs.drop(). I can also easily store the marked trials for each subject for quick reanalysis by removing the same trials. This makes the preprocessing transparent and easy to replicate.

Additionally, I sometimes look at individual epochs with epochs.plot() and mark trials for rejection. I would like to be able to access these trials marked as bad and add them to my list of trials to reject for the current subjects instead of automatically removing them after closing the epochs plot.

#### Describe the problem
After marking bad trials in the epochs.plot() and closing the window, the trials get automatically dropped.

#### Describe your solution
I want it to just store marked trials in a similar way as it stores channels marked as bad.

jasmainak, 2019-05-22:
This is [possible in autoreject](http://autoreject.github.io/auto_examples/plot_visualize_bad_epochs.html#sphx-glr-auto-examples-plot-visualize-bad-epochs-py). However, that said it seems it's oft-requested feature so I would not be opposed to copying the small code from autoreject into MNE. So the autoreject code is not calling a bunch of private functions from MNE which is hard to maintain in the long run.

mmagnuski, 2019-05-22:
That's a call for Epochs Annotations! :)

agramfort, 2019-05-23:
do you know about the epochs.drop_log to keep track of the epochs that have
been dropped and why?

>

mmagnuski, 2019-05-23:
If I remember correctly `.drop_log` is not very human-friendly.  :)

agramfort, 2019-05-23:
epochs.plot_drop_log helps?

>

Eklund89, 2019-05-23:
@agramfort I thought the plot_drop_log was a visual tool, correct? I need the trial numbers so i can save them to a logfile.

larsoner, 2019-05-23:
You can iterate over the drop_log list, any non empty entry has been dropped. Or use epochs.selection in combination with the length of your original events list.

mmagnuski, 2019-05-23:
@agramfort I have to correct my statement about `.drop_log` not being human-friendly. I checked and `.drop_log` with the help of `.selection` does exactly what is requested here.

mmagnuski, 2019-05-23:
@Eklund89 Maybe you would be willing to submit a PR with example showing how to access these information using `.drop_log` and `.selection` - I guess there are not much information on these in the tutorials/examples.