mne-tools/mne-python #6363: DOC/BUG: Old intro docs

larsoner, 2019-05-22:
https://mne-tools.github.io/mne-python-intro/ still exists. Should we update or remove?

larsoner, 2019-05-22:
(thanks @xexeo for reminding us it's there)

drammock, 2019-05-22:
I've been wanting to remove it, but I don't know what pages (possibly outside of MNE) link to it (and hence would die if we removed the page). Do you think it's worth setting up a redirect to https://mne-tools.github.io/dev/auto_tutorials/intro/plot_introduction.html ?

larsoner, 2019-05-22:
+1 for redirecting to that tutorial. It should be easy to do via a PR to `mne-tools.github.io` repo. 

Also I just noticed that that tutorial does not seem to be linked to from https://mne-tools.github.io/dev/documentation.html. Seems like it should live under "MNE basics", no?

drammock, 2019-05-22:
yeah, I'll add it there and set up the redirect. I was lazy about adding to `documentation.html` because I know that page will get an overhaul soon.

drammock, 2019-05-22:
so, I admit confusion.  `git grep mne-python-intro` shows several links to that page, showing up in versions `0.11` through `0.17` (but not in `dev`), and the page itself doesn't seem to be part of the repo (?) which makes it hard for me to figure out how to replace it with a redirect.  Also various `git grep` calls on short spans of text found in https://mne-tools.github.io/mne-python-intro/ return no hits.

larsoner, 2019-05-22:
Someone probably added the html and images manually. There was a time back
when none of this stuff was automated...

So you can just make a PR to that repo to remove the files file and replace
them with a single redirect html file. If this is not clear I can do it too.

>

drammock, 2019-05-22:
It's clear, it's just that I can't even *find* the file I'm supposed to replace with a re-direct.  If it's not in the `mne-tools.github.io` repository, where else can it be?

larsoner, 2019-05-22:
Ahh GitHub trickery.

https://github.com/mne-tools/mne-python-intro/tree/gh-pages

We can modify this branch to just have a redirect.

drammock, 2019-05-22:
cool, thanks for helping me find that

drammock, 2019-05-23:
@larsoner should we still add the redirect page?  I guess it would go in `doc/mne-python-intro/index.html`?

larsoner, 2019-05-23:
Yes please open a PR for it

drammock, 2019-05-23:
closed by mne-tools/mne-tools.github.io#16

closed 2019-05-23 by drammock