mne-tools/mne-python #6364: website repo size is getting out of hand

drammock, 2019-05-22:
the repo for `mne-tools.github.io` is now 4.5 GB (for comparison, the MNE-Python repo is 0.95 GB).  This makes it unweildy to change anything; the cloning process can take several minutes. @larsoner suggested that we open an issue here to collect opinions / ideas about whether it's worth it to try to shrink the website repo, and if so, how to go about it in the least disruptive way.

jasmainak, 2019-05-22:
why don't you squash the commits?

larsoner, 2019-05-22:
@jasmainak yes we need to rewrite history, squashing would be a good way to do it. But rewriting history is dangerous so we have to be careful about it and make sure people who would be affected are on board.

The things I foresee this potentially breaking are:

1. CircleCI when it caches the `mne-tools.github.io` repo in order to update it.
2. martinos sync scripts
3. If there is some old doc that existed temporarily and we realize later we wanted it, we can't get it back.

The first two currently use caching (implicitly or explicitly) of repos and `git pull` to update. These can be fixed by doing a `git fetch` followed by `git reset --hard upstream/master` instead of a simple `git pull`, so I don't see them as blockers.

As far as losing things (3), I thought about squashing to one commit per release period (or even really one every 6 months or year) as good enough. While that seems a bit overkill since we store our old doc versions in our new doc pages, if we accidentally remove and force-push-squash, we could lose them. So I propose:

1. One commit per release, i.e. one for 0.11, 0.12, ... 0.18.
2. We continue to let CircleCI push for every `master` and `maint` update.
3. Make it part of the release process to squash the history between the previous release and the current one.

drammock, 2019-05-22:
:+1: for squashing to 1 commit per release for old releases.

:+1: for letting CircleCI push commits during a dev cycle.

:+1: for making website commit squashing part of the release process.

agramfort, 2019-05-23:
agreed

martinos sync scripts do clone with depth 1 so don't need history

>

larsoner, 2019-05-23:
Okay I'm going to see if I can do this. 1200+ commits will hopefully get squashed to ~10 today...

larsoner, 2019-05-23:
Fixed:

https://github.com/mne-tools/mne-tools.github.io/commits/master

Size after a fresh `git clone` is 1.6GB, so still big, but not as bad.

closed 2019-05-23 by larsoner

drammock, 2019-05-23:
Thanks @larsoner!