mne-tools/mne-python #6365: DOC: Installation troubleshooting

larsoner, 2019-05-22:
Talking to @jasmainak about an upcoming workshop and getting emails, I think we would benefit from a dedicated page about troubleshooting installation issues. A page like this could be nice:

http://ryan-leung.github.io/PHYS4650_Python_Tutorial/installing-on-linux.html

Basically something with green/checkboxes saying "this is good" and red/X's saying "this is bad" at different stages, with links to "how to fix bad" would be helpful. I'm not sure how it should be structured. @drammock do you want to try, or would you rather I take a stab at it?

Maybe this should be broken into:

1. Ensure conda is installed and accessible from your terminal
2. Ensure the `mne` environment is active
3. Ensure the dependencies are installed (`sys_info`, maybe `mlab.test_plot3d`)
4. Ensure the dependencies work (esp. `mayavi`, with interactivity / OpenGL issues)
5. Ensure Spyder works (?) if we want to push people that way

Example diagnostics that should appear somewhere, with what they imply:

- `which conda` (good: correct path, bad: no output)
- `conda 
- `which python` (good: Anaconda, bad: anything else e.g. system python)
- `which pip` (good: Anaconda pip, bad: anything else e.g. system pip)
- `echo $PATH` (good: Anaconda on path; bad: not on path)
- `python -c "import mne"` (good: works, bad: fails)
- `echo $PYTHONPATH` (good: empty, bad: could point somewhere)

Other things:
- tcsh vs bash (
- old way of doing conda (PATH modification) vs new way (sourcing their script); the corresponding `conda activate` vs `source activate`

If others have installation tips let me know and we'll try to add them.

drammock, 2019-05-22:
depends on when your workshop is; I likely can't get to this until next week.

larsoner, 2019-05-22:
Okay cool. I'll see if I can do it tomorrow morning. Hopefully I can get most content in this week, and you can work some readability / incorporation magic on it :)

jasmainak, 2019-05-22:
I like the idea of giving a list of diagnostic commands with a good and bad. It's how I usually troubleshoot myself. another command I would add to the list is:

```py
$ pip list | grep mne
```

This is really useful to figure out which version of mne is being used and if it's the latest one

drammock, 2019-05-22:
just a note to say that there is a vestigial bit of the installation docs [here](https://mne-tools.github.io/stable/install_mne_python.html#troubleshooting-mne-python-installation) about troubleshooting. That should be revised once there is a dedicated troubleshooting page.

PR #6393 (2019-06-01): MRG, ENH: Add some troubleshooting to install docs
Closes #6365.
closed 2019-06-01 by larsoner