mne-tools/mne-python #6367: Incorrect scaling on macOS HiDPI when resizing plot window

cbrnr, 2019-05-23:
We've already fixed the incorrectly scaled `plot_raw` window on macOS on a HiDPI (retina) display in #6043 (discussion in #6034). However, the same issue occurs after resizing the `plot_raw` window. After resizing, the axes overlap again. Is there another place where we calculate the layout of this figure? Could other plots also be affected?

larsoner, 2019-05-23:
`git grep resize` shows (among other things):
```
mne/viz/raw.py:    callback_resize = partial(_helper_raw_resize, params=params)
```
Can you look?

cbrnr, 2019-05-24:
`_helper_raw_resize` calls `_layout_figure`, which works on initialization but not on resize. It's hard to debug for me, because while debugging it doesn't even work on initialization...

jona-sassenhagen, 2019-05-28:
@stfnrpplngr any ideas?

larsoner, 2019-05-28:
I cannot reproduce on my HiDPI OSX machine. Maybe it's a PyQt / `matplotlib` bug.

cbrnr, 2019-05-29:
It only affects the `osx` backend - I don't have this issue with the `qt5` backend. @larsoner can you reproduce with `osx`?

larsoner, 2019-05-29:
I can't (currently at least) test it because I get the "ImportError: Python is not installed as a framework" error.

cbrnr, 2019-05-29:
Matplotlib 3.1.0 should have fixed this issue: https://matplotlib.org/3.1.0/users/whats_new.html#matplotlib-no-longer-requires-framework-app-build-on-macosx-backend

larsoner, 2019-05-29:
Ahh okay I can try on Friday

larsoner, 2019-05-29:
... but also this seems like it's almost certainly a `mpl` problem, not an MNE one, since `Qt5Agg` works but `macOS` does not. Can we close here and you can open a MPL issue after making a minimal example (once I confirm I have the same problem later this week)?

cbrnr, 2019-05-29:
Yes, I agree it should be a matplotlib issue. Can you help me make a minimal example?

larsoner, 2019-05-30:
You can try writing a few lines of code that will:

1. Create figure
2. Add subplot axes (using whatever `plt` interface we use in `raw.plot` under the hood)
3. Connect resize event handler
4. Show it works with Qt5 backend, fails with OSX backend

If step (4) does not work, then we must have some additional complexity in our code which breaks things, and we can instead try working from the "top down" (removing bits of our code until things work properly) instead of the "bottom up" (creating MPL code from scratch that works until we add something that breaks).

PR #6395 (2019-06-01): MRG, BUG: Fix MacOSX backend HiDPI scaling
Closes #6367.

@cbrnr can you see if this fixes things for you? In case you're curious, I found it just by creating a MacOSX figure and inspecting `fig.canvas._` properties (actually I looked right at `fig.canvas._d`-tab-tab and got lucky that the MacOSX private propetry `_device_scale` started with the same letter as the Qt5Agg one `_dpi_ratio`).

Upstream report https://github.com/matplotlib/matplotlib/issues/14405
closed 2019-06-01 by cbrnr