mne-tools/mne-python #6375: [BUG] Concatenate epochs with different electrodes order

fraimondo, 2019-05-27:
#### Describe the bug
When concatenating epochs, channel order is not considered as a difference. So epochs with different channel order are concatenated without warning/error.

From _compare_epochs_infos in mne/epochs.py:
```Python
    if set(info1['ch_names']) != set(info2['ch_names']):
        raise ValueError('epochs[%d][\'info\'][\'ch_names\'] must match' % ind)
```

#### Steps and/or code to reproduce
```Python
import mne

ch_names_1 = ['EEG1', 'EEG2']
info1 = mne.create_info(ch_names_1, 250., ch_types='eeg')
data1 = np.concatenate([np.ones((3, 1, 4)), np.ones((3, 1, 4)) * 2], axis=1)

ch_names_2 = ['EEG2', 'EEG1']
info2 = mne.create_info(ch_names_2, 250., ch_types='eeg')
data2 = np.concatenate([np.ones((3, 1, 4)) * 2, np.ones((3, 1, 4))], axis=1)

epochs1 = mne.EpochsArray(data1, info1)
epochs2 = mne.EpochsArray(data2, info2)

epochs = mne.concatenate_epochs([epochs1, epochs2])
```
In this example, the epochs channels have inverse order. But the order kept is from `epochs1`. So data from electrode EEG2 gets appended to EEG1.

#### Expected results
A warning or an error.

#### Actual results
Nothing

agramfort, 2019-05-27:
for me it should just work.

>

fraimondo, 2019-05-27:
I don’t disagree with that, as long as there is documentation. I had this doubt and I had to revise the code and make the test to check if different electrode order would make a difference or not.

As its stand now, the definition of equality between info objects is not strict.

agramfort, 2019-05-27:
do you have time to open a PR?

>

jona-sassenhagen, 2019-06-03:
See these two functions: https://github.com/mne-tools/mne-python/blob/b9b2c6884dc663ef42338bdccc9e7838820b29b3/mne/evoked.py#L834

combine_evoked and the helper before.

KSuljic, 2020-07-01:
This should fix this, right?

https://mne.tools/stable/generated/mne.channels.equalize_channels.html#mne.channels.equalize_channels

hoechenberger, 2021-02-14:
Yes, `mne.channels.equalize_channels` should do the job.

closed 2021-02-14 by hoechenberger