mne-tools/mne-python #6377: Amplitude of interpolated channels very small (part 2)

jorpelga7, 2019-05-27:
Hi!
I am trying to interpolate dead channels in the mode described below, but the interpolated channels seem to come out with negligible amplitude (i.e. they look flat). In a related thread, I read that specifying origin = (0., 0., 0.01) in raw.interpolate_bads() should do the trick, but it still doesn't seem to work when I try. I'd very much appreciate some help with this.
Thanks in advance!
Joan

import mne
raw_fname = '/path/to/file'
raw = mne.io.Raw(raw_fname, preload=True)
data = raw.get_data()

# mark bads on plot
raw.plot(block=True)
raw.interpolate_bads()

jasmainak, 2019-05-28:
Can you share your data?

agramfort, 2019-05-28:
can you show the layout of your channels? How many channels do you have?
are the channels you try to intepolate far?

jorpelga7, 2019-05-28:
Hi,
here is a sample file. 
https://drive.google.com/file/d/15JzRPtAKpKV4oz08eMzY4N-pVhgPwYcE/view?usp=sharing

I basically do:

import mne
fname = 'path to ... /R1_SyllBlock.sqd'
raw = mne.io.read_raw_kit(fname, preload=True, allow_unknown_format = True)
raw.info['bads'] = ['MEG 026', 'MEG 041', 'MEG 065', 'MEG 066', 'MEG 088', 'MEG 097', 'MEG 153']

raw.interpolate_bads()
mne.viz.plot_raw(raw)

Thanks!

agramfort, 2019-05-29:
I don't understand the problem. I ran your code and signals look fine.

for example 'MEG 066' has proper ampitudes.

>

jorpelga7, 2019-05-29:
And you did not add anything at all to the code?
Could it be something with the version of mne/python I am using? (I'm using Python 3.6.8, and MNE 0.17.2)
When I run it the bads are restored so that raw.info['bads'] = [], but they look flat in the plot...

larsoner, 2019-05-29:
It's possible we fixed a bug in the last 6 mo between 17.2 and the recent 18.1 release, can you upgrade MNE and see if it works better?

jorpelga7, 2019-05-30:
Unfortunately it still does not work... I just spoke to a colleague here and he has had the same issue without being able to solve it for now :(
In any case, thanks very much for looking into it btw

drammock, 2019-05-30:
I cannot replicate either:
```python
import mne
fname = 'R1_SyllBlock.sqd'
raw = mne.io.read_raw_kit(fname, preload=True, allow_unknown_format=True)
bads = ['MEG 026', 'MEG 041', 'MEG 065', 'MEG 066', 'MEG 088', 'MEG 097', 'MEG 153']
raw.info['bads'] = bads
orig_raw = raw.copy()
raw.interpolate_bads()
for obj in (raw, orig_raw):
    obj.reorder_channels(bads)
    obj.plot()
```

[image]

```python
>>> mne.sys_info()                                                                                                                  
Platform:      Linux-4.15.0-50-generic-x86_64-with-debian-buster-sid
Python:        3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44)  [GCC 7.3.0]
Executable:    /opt/miniconda3/envs/mnedev/bin/python
CPU:           x86_64: 8 cores
Memory:        7.6 GB

mne:           0.19.dev0
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.2 {backend=Qt5Agg}

sklearn:       0.20.2
nibabel:       2.3.1
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.23.4
dipy:          0.15.0
```

jorpelga7, 2019-05-30:
Gosh...
This is what I get with the same exact code:

[image]

[image]

agramfort, 2019-05-30:
can you try on a fresh conda environment?

larsoner, 2019-05-30:
I can reproduce:
```
Qt WebEngine seems to be initialized from a plugin. Please set Qt::AA_ShareOpenGLContexts using QCoreApplication::setAttribute before constructing QGuiApplication.
Platform:      Darwin-18.6.0-x86_64-i386-64bit
Python:        3.6.8 |Anaconda custom (x86_64)| (default, Dec 29 2018, 19:04:46)  [GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/larsoner/anaconda3/bin/python
CPU:           i386: 4 cores
Memory:        8.0 GB

mne:           0.19.dev0
numpy:         1.16.4 {blas=openblas, lapack=openblas}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
nibabel:       2.4.0dev
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.11.3}
cupy:          Not found
pandas:        0.24.2
dipy:          0.14.0
```
One consistent difference between those who can and cannot is the NumPy version. Perhaps they changed some handling of ints/floats. I'll look.

larsoner, 2019-05-30:
Oh wait, no I can't reproduce. I mis-interpreted the plots...

jorpelga7, 2019-05-30:
Don't know exactly what it is, but I deleted the mne folder under envs in anaconda, created a new environment altogether as **agramfort** suggested, and it worked! 
Before that, I realized that simply running the code in the terminal after user$ ipython (i.e. outside the conda env) was also ok, but initializing the env by user$ conda activate mne before running ipython did not work...
Thank you very much! let me know if there is any info I could provide to help identify the precise cause.

larsoner, 2019-05-30:
Not sure what it could be. In the absence of more information (that is now probably deleted from your disk :) ) we'll close for now. But if this comes up again we can try to figure out what was going on.

closed 2019-05-30 by larsoner