mne-tools/mne-python #6378: Some remarks to simulate_raw

makkostya, 2019-05-28:
After working with simulation module I have some remarks:

1. Maybe we should add "exclude=bads" to channel selection:
`meeg_picks = pick_types(info, meg=True, eeg=True, exclude=[])`

I  faced a problem when tried to simulate data in the project, where I already computed `fwd` which didn't have bad channels  `['MEG 2443', 'EEG 053']`. But these channels are included when I  simulate raw and as a result I have an assertion error in `simulate_raw`:
`assert fwd['sol']['data'].shape[0] == len(meeg_picks)`

2. If my info does not contain stim channel maybe it worth to add it automatically in `simulate_raw`. Cause now it results in an error and msg that there is no stim channel.

larsoner, 2019-05-28:
> Maybe we should add "exclude=bads" to channel selection:

I think I'd rather have it simulate data even for the bad channels without error. For now you can work around it by setting `info['bads'] = []` before calling `simulate_raw`.

> If my info does not contain stim channel maybe it worth to add it automatically in simulate_raw. Cause now it results in an error and msg that there is no stim channel.

I think it should simulate without adding a stim channel without error. If people want a stim channel, they should add it to their template raw instance or info (using e.g. raw.add_channels).

makkostya, 2019-05-28:
> I think it should simulate without adding a stim channel without error. If people want a stim channel, they should add it to their template raw instance or info (using e.g. raw.add_channels).

It shows error when I try to call `events = mne.find_events(raw)` . I think that as far as `events` is an input to simulate_raw, the stim_channel should be automatically added. I don't see why user should bother to manually add  `raw.add_channels`. He may not even know that it exists, but he still may want to compute evoked from raw in a fast way by using `mne.find_events`

larsoner, 2019-05-28:
>  I don't see why user should bother to manually add raw.add_channels. He may not even know that it exists, but he still may want to compute evoked from raw in a fast way by using mne.find_events

The idea is that the raw template you provide should contain what you want to simulate. If it has MEG, it does MEG. If it has EEG, it has EEG. If it has stim channels, it will contain stim channels, etc. Perhaps it's worth breaking this from a convenience perspective, but I'd still slightly prefer to get people to think about what they want simulated, and maybe `warn` if there is no stim channel instead. There are scenarios where no stim channel would be necessary, too, for example simulating a sleep-stage recording.

agramfort, 2019-05-28:
can you show the code that you expected to work and which did not?

>

makkostya, 2019-05-28:
This piece of code would simulate raw data without stim channel if evoked does not have one (even if source_simulator contains events):
`raw_region = mne.simulation.simulate_raw(evoked.info, source_simulator, forward=fwd_fixed)`
It's not really a problem, I still can get evoked from raw if I have events array. For me it is just seems counterintuitive that events information is not preserved in raw data. I would expect to be able to get events from raw, even if I hadn't the original events array:
`events = mne.find_events(raw_region)`

agramfort, 2019-05-28:
the reason I ask to see the code is because it's faster to read than
text and it allows
to see if the code user have to write is actually counter intuitive.

NeuroLaunch, 2021-03-16:
I'm planning to initiate a new Issue for fixing/enhancing raw simulations during the 2021 MNE Sprint.  I can incorporate @larsoner's suggestion to give a `warn` if channel setups between original raw, events, etc, are incongruous.  Please let me know if there is anything else from this issue, or related, that I should include.  (I'll follow-up with the new Issue # when I can.)