mne-tools/mne-python #6382: BUG: `inverse_sparse` with `weights_min=None`

larsoner, 2019-05-29:
Doing:
```
stc_mxne = mixed_norm(
    evoked.copy().crop(0, 0.4), fwd, cov, alpha,
    weights=stc_dspm, weights_min=8., active_set_size=10,
    n_mxne_iter=n_mxne_iter, verbose=True)
```
is fine, but removing `weights_min=8.` (which takes it to the default `None`) yields:
```
  File "/Users/larsoner/python/mne-python/mne/inverse_sparse/mxne_inverse.py", line 358, in mixed_norm
    weights, weights_min)
  File "/Users/larsoner/python/mne-python/mne/inverse_sparse/mxne_inverse.py", line 65, in _prepare_gain
    forward, gain, source_weighting, weights, weights_min)
  File "</Users/larsoner/python/mne-python/mne/externals/decorator.py:decorator-gen-372>", line 2, in _prepare_weights
  File "/Users/larsoner/python/mne-python/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/Users/larsoner/python/mne-python/mne/inverse_sparse/mxne_inverse.py", line 29, in _prepare_weights
    if weights_min > weights_max:
TypeError: '<' not supported between instances of 'float' and 'NoneType'
```

larsoner, 2019-05-29:
@agramfort I see a few options, not sure which is preferable:

- enforce that if someone passes `weights` they must pass `weights_min`
- if `weights_min is None` then use some automatic criterion (e.g. 50th percentile / include half of all verts?)
- set `weights_min=0.` by default (but then if someone passes `weights` it will be effectively ignored, so not a great option)

agramfort, 2019-05-29:
weights_min=0. is fine with me

PR #6384 (2019-05-29): MRG, FIX: weights_min in mixed_norm
Closes  #6382.
closed 2019-05-29 by larsoner