mne-tools/mne-python #6383: AttributeError: 'EpochsArray' object has no attribute 'drop_epochs'

talhaanwarch, 2019-05-29:
I am trying to remove epoch from epochsArray object. but i am getting AttributeError: 'EpochsArray' object has no attribute 'drop_epochs'.
 Here is the plot image..
http://prntscr.com/nupz7m

Or is there any way to remove segment from above image by marking them bad

agramfort, 2019-05-29:
it seems indeed a bug. Can you share a snippet to replicate the full problem
using the sample data?

talhaanwarch, 2019-05-29:
ok, i am not perfect in python. here i try to upload the sample data and file
https://github.com/talhaanwarch/EEG-Shizophrenia

agramfort, 2019-05-29:
I cannot replicate the problem with:

```
import mne

from mne.datasets import sample

reject = dict(grad=4000e-13, mag=4e-12, eog=150e-6)

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname)

events = mne.find_events(raw, stim_channel='STI 014')
event_id = {"auditory/left": 1}
tmin = -0.2
tmax = 0.5
baseline = (None, 0)
picks_meg = mne.pick_types(raw.info, meg=True, eeg=False, eog=True,
                           stim=False, exclude='bads')
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    picks=picks_meg, baseline=baseline, reject=None)
print(epochs)

epochs = mne.EpochsArray(epochs.get_data(), epochs.info, tmin=tmin)
epochs.plot()
```

even with an EpochsArray  I can remove interactively epochs etc.

closed 2019-09-11 by larsoner

jessicaloke, 2019-12-12:
Not sure if this was followed up - but instead of drop_epoch(indices) it should just be drop(idices). That's how it's defined in the original function.