mne-tools/mne-python #6389: Cannot read some EDF files, raises ValueError: tmin should be positive

guiweber, 2019-05-30:
I am using the latest MNE V0.19dev to read EEG data from the CHB-MIT database, located at https://alpha.physionet.org/content/chbmit/1.0.0/

Some files can be opened just fine with `mne.io.read_raw_edf` but other raise `ValueError: tmin should be positive.`

This data set has been used extensively in the literature, so I presume the files should be usable. Also, the traceback shows that the error originates in crop, called by set_annotations, which indicates that the problematic values are in annotations, which shouldn't prevent using the data.

#### Steps to reproduce
1. Download https://alpha.physionet.org/static/published-projects/chbmit/1.0.0/chb06/chb06_03.edf
1. Try opening the file with `mne.io.read_raw_edf`

#### Expected results
If the file parameters or annotations are wrong but the data is otherwise ok, the parameters/annotations should be set to reasonable default values and a warning should be produced. It should still be possible to load the file.

#### Actual results
It is impossible to load the file

#### Debug output and traceback 
```Python
raw = mne.io.read_raw_edf(file, preload=True, verbose='DEBUG')
```
Extracting EDF parameters from xxx\chb06_03.edf...
EDF file detected
xxx/main.py:19: RuntimeWarning: Channel names are not unique, found duplicates for: {'T8-P8'}. Applying running numbers for duplicates.
Setting channel info structure...
  FP1-F7: range=1600.0 cal=4095.0
  F7-T7: range=1600.0 cal=4095.0
  T7-P7: range=1600.0 cal=4095.0
  P7-O1: range=1600.0 cal=4095.0
  FP1-F3: range=1600.0 cal=4095.0
  F3-C3: range=1600.0 cal=4095.0
  C3-P3: range=1600.0 cal=4095.0
  P3-O1: range=1600.0 cal=4095.0
  FP2-F4: range=1600.0 cal=4095.0
  F4-C4: range=1600.0 cal=4095.0
  C4-P4: range=1600.0 cal=4095.0
  P4-O2: range=1600.0 cal=4095.0
  FP2-F8: range=1600.0 cal=4095.0
  F8-T8: range=1600.0 cal=4095.0
  T8-P8-0: range=1600.0 cal=4095.0
  P8-O2: range=1600.0 cal=4095.0
  FZ-CZ: range=1600.0 cal=4095.0
  CZ-PZ: range=1600.0 cal=4095.0
  P7-T7: range=1600.0 cal=4095.0
  T7-FT9: range=1600.0 cal=4095.0
  FT9-FT10: range=1600.0 cal=4095.0
  FT10-T8: range=1600.0 cal=4095.0
  T8-P8-1: range=1600.0 cal=4095.0
Creating raw.info structure...
Reading 0 ... 3686399  =      0.000 ... 14399.996 secs...
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.2\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.2\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "xxx/main.py", line 19, in <module>
    raw = mne.io.read_raw_edf(file, preload=True, verbose='DEBUG')
  File "C:\Program Files\Python37\lib\site-packages\mne\io\edf\edf.py", line 1244, in read_raw_edf
    verbose=verbose)
  File "<C:\Program Files\Python37\lib\site-packages\mne\externals\decorator.py:decorator-gen-152>", line 2, in __init__
  File "C:\Program Files\Python37\lib\site-packages\mne\utils\_logging.py", line 88, in wrapper
    return function(*args, **kwargs)
  File "C:\Program Files\Python37\lib\site-packages\mne\io\edf\edf.py", line 193, in __init__
    description=desc, orig_time=None))
  File "C:\Program Files\Python37\lib\site-packages\mne\io\base.py", line 795, in set_annotations
    emit_warning=emit_warning)
  File "C:\Program Files\Python37\lib\site-packages\mne\annotations.py", line 361, in crop
    raise ValueError('tmin should be positive.')
ValueError: tmin should be positive.

#### Additional information

Platform:      Windows-10-10.0.17763-SP0
Python:        3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)]
Executable:    xxx\venv\Scripts\python.exe
CPU:           Intel64 Family 6 Model 94 Stepping 3, GenuineIntel: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.19.dev0
numpy:         1.16.1 {blas=C:\\projects\\numpy-wheels\\numpy\\build\\openblas, lapack=C:\\projects\\numpy-wheels\\numpy\\build\\openblas}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=TkAgg}
sklearn:       0.21.1
nibabel:       Not found
mayavi:        Not found
cupy:          Not found
pandas:        0.24.2
dipy:          Not found

larsoner, 2019-05-30:
@massich can you look?

cbrnr, 2019-05-31:
The file does not have any annotations. The problem is likely the start of recording data, which is set to 2060-01-27 03:09:42.

cbrnr, 2019-05-31:
Or rather, 1960 instead of 2060 (2060 is returned by biosig, 1960 by Python or R datetime functions).

agramfort, 2019-06-02:
I confirm that #6391 fixes the bug.

shall we merge and backport to 0.18 branch?

closed 2019-06-03 by agramfort

massich, 2019-06-03:
Yes, #6391 fixes the bug, and I guess that we could backport it. I'll do it today

Matteo73, 2019-06-03:
@massich 
Thanks for backporting it to 0.18. When the 0.18 updated version will be avaible?

Cheers,
Matteo

agramfort, 2019-06-03:
can you rely on master for your work?

>

Matteo73, 2019-06-04:
@agramfort 
Yes, of course, I can. I will migrate to the stable version when it is available.

Cheers,
Matteo

PR #6391 (2019-06-03): Fix error when checking annotations with negative timestamps
Fixes #6389.