mne-tools/mne-python #6390: RFC: confusing API for make_fixed_length_events?

drammock, 2019-05-30:
The API for `mne.make_fixed_length_events` is confusing (at least to me). The main problem, I think, is that it has a `duration` parameter but it makes an events array that is exclusively made up of instantaneous events (given the presence of a `duration` parameter I would have expected something more like the result of `mne.find_events(..., ouptut='step')`). Also the `overlap` parameter is superfluous, since you can get the same result by either passing `overlap=n, duration=m` or `overlap=0, duration=m-n`.

GIven what I think is the typical use case (people want instantaneous events at a fixed spacing), I think a more sensible API is this:

- change name to `mne.make_equally_spaced_events`
- change parameter `duration` to `spacing`
- remove parameter `overlap`

So new function signature would be `mne.make_equally_spaced_events(raw, id=1, start=0., stop=None, spacing=1., first_samp=True)`

Then if users want overlapping events, they get it by specifying the right `tmin`, `tmax` in the Epochs constructor.

larsoner, 2019-05-31:
The variables are named this way to match with the typical use cases, for example Welch PSD of continuous data broken into chunks. In this domain you usually specify duration and overlap to be used, not the net spacing. In MNE you give these to the function, then tmin=0 and tmax=duration when epoching and you're done.

So for me -1 not worth a deprecation cycle to change this stuff, or get a better name. The headache for users for a dep cycle (which we might already do too often) is greater than the net pain from confusion of our function and argument naming. I'd rather just document the behavior.

agramfort, 2019-06-02:
I am also -0.5 on a deprecation cycle here. A better docstring will
surely help though

drammock, 2019-06-02:
Ok I'll work on the docstring

jona-sassenhagen, 2019-06-03:
funny, I just wanted to open an issue for make_fixed_length_events.

My issue: `id` is a built-in that we are shadowing here.

larsoner, 2019-06-03:
Agreed we should not shadow it and if I were rewriting the function from scratch I would do `id_` but not worth the deprecation to change it now

drammock, 2019-06-03:
I'm not convinced by the "we deprecate too often / it's a headache for users" argument.  MNE is 0.XX software. Changes should be expected, and users can / should tie their analyses to particular versions.

jasmainak, 2019-06-03:
> I'm not convinced by the "we deprecate too often / it's a headache for users" argument

I invite you to have lunch with my friends at MGH :) People want their scripts to run even 3 to 5 years later -- we're nowhere close but some stability in terms of API will not hurt.

drammock, 2019-06-03:
If they tie each project to a particular version, then their scripts *will* run 3-5 years later.  But regardless of that, stability != stagnation.  I'm fine with a long-ish deprecation cycle, if that's what is needed for this community.  But leaving in arg names that shadow built-ins as `wontfix` seems crazy to me.

jona-sassenhagen, 2019-06-03:
I'm for once in agreement with Mainak, and for being conservative ... the shadowed builtin is local to a small function where it is safely contained. API stability allows people to upgrade and get bugfixes (we don't backport much!) while having guarantees their scripts won't break.

Maybe with a reasonable deprecation cycle, but it seems somewhat low priority to me too.

jasmainak, 2019-06-03:
> If they tie each project to a particular version, then their scripts will run 3-5 years later.

I tried this approach but it doesn't work because most people want their scripts to run 3-5 years later _because_ they want to use a new feature from a new version. That is to say, it's usually because they want to try something cool and new on their old data, not because they want their scripts to produce the same results as it did 3-5 years ago.

drammock, 2019-06-03:
I still do not agree, but I yield.

closed 2019-06-03 by drammock