mne-tools/mne-python #6406: Auto-scaling plot.raw in combination with remove_dc and/or filter

cbrnr, 2019-06-04:
I often work with EEG data recorded without a highpass filter, which means that channels have large offsets different from zero. In a first step, I usually want to create a quick `raw.plot` to look at the raw data. Unfortunately, using `raw.plot(scalings='auto', remove_dc=True)` generally produces flat lines. This is despite setting `remove_dc=True`, the `scalings` are still inferred from the original data containing large offsets. Here's a short example to illustrate what I mean:

```
import numpy as np
import mne

data = np.random.normal(loc=500, scale=1, size=(10, 10000))
info = mne.create_info(ch_names=data.shape[0], sfreq=100)
raw = mne.io.RawArray(data, info)
raw.plot(scalings="auto", remove_dc=True)
```

[image]

It would be really useful if `scalings` were determined from the offset-corrected values, which would result in the following plot (simulated with `np.random.normal(loc=0, scale=1, size=(10, 10000))` here):

[image]

The same is true for the `highpass` and `lowpass` arguments. It would be useful if the automatic scalings were calculated on the filtered signals. Otherwise, I have to supply manually chosen values for `scalings` in order for the plot to actually show the signals (e.g. `raw.plot(scalings=2, remove_dc=True)` or channel type specific `raw.plot(scalings={'misc': 2}, remove_dc=True)`), because the default e.g. for EEG channels is not appropriate for many EEG signals I use.

What do you think? If this is a useful change, I'd be happy if someone could give me a pointer to where this should be implemented. Currently, we compute automatic scalings at the very start of `raw.plot`, but `remove_dc` and filtering is implemented in the `_update_raw_data` function which works only on the currently visible data.

agramfort, 2019-06-04:
+1 for autoscaling after filtering/dc correct

>

larsoner, 2019-06-04:
For filtering you'll need to load all data into memory, filter it, and compute.

For DC correction, the operation is done *on your current view of the data*. But we probably still get a good value by (hopefully quickly) using `make_fixed_length_events`, creating Epochs from this baselined with `(None, None)`, then getting the data and computing the scaling factors from that.

cbrnr, 2019-06-04:
What if we add a new autoscale feature which works based on the current view of the data only?

larsoner, 2019-06-04:
To me it seems like a misleading (and challenging) way to look at the data to have the scale change as a function of which time window you've cropped to. If the primary motivation is just to avoid the technical challenge of loading into memory it doesn't seem like it's worth it.

cbrnr, 2019-06-04:
But don't both filter and remove_dc arguments to `raw.plot` work only on the currently visible data segment? DC can vary tremendously if there is a large drift and you calculate the global mean vs. the mean within the visible segment.

larsoner, 2019-06-04:
Yes they both operate only on the given segment. Removing frequency content (what they do) on a per-window basis is different from changing the effective scale bar / pixel-to-units scaling on a per-window basis (what you propose).

cbrnr, 2019-06-04:
Yes, I'm getting ahead of myself. Both options would be useful, but I'm OK with tackling the scaling based on the whole signal first. However, how do we consolidate `remove_dc` with this approach then? One is based on the whole signal while the other is based on the visible segment...

larsoner, 2019-06-04:
>  how do we consolidate remove_dc with this approach then?

See above -- construct an `Epochs` object with `baseline=(None, None)` using events from `make_fixed_length_events`, get the data from it, (effecitively) concatenate the epochs to be raw-like `(n_channels, n_times)`, and use this to compute the scale factors (using `std` or however we currently do it for the original data, and will do it for the filtered data)

cbrnr, 2019-06-04:
Which epoch duration would you use?

larsoner, 2019-06-04:
The (starting) `raw.plot` duration

cbrnr, 2019-06-05:
OK, I got myself into circular dependency hell. I'm trying to use `mne.make_fixed_length_events`, but if I do:

```
from .. import make_fixed_length_events
```

at the top of the module I get

```
ImportError: cannot import name 'make_fixed_length_events' from 'mne' (/Users/clemens/Repositories/mne-python/mne/__init__.py)`
```

It works if I do

```
from ..event import make_fixed_length_events
```

However, I have other imports that I need such as `RawArray` and `create_info`, and there is no way I can import these at the top without getting an `ImportError`.

Is it OK if I just import these functions within the function I need them? I know there are several approaches how to handle this situation, so I wanted to know what you guys preferred.

cbrnr, 2019-06-05:
Turns out it is much easier to achieve a segment-based offset correction using plain old numpy, so I don't need the additional imports after all. I'm still interested on how you handle such cases though.

Also, it depends a lot on the actual data whether or not this new autoscale works well. For the toy example, I get a nice output:

[image]

But with one of my real-world EEG data sets, I still don't really see the signals, but the scaling factor determined by the autoscale procedure did improve significantly:

Old auto-scale: 0.2596253484038322
New auto-scale: 0.004032760298876942

What I'd consider a good value (where I can see the actual variations in the signals): 0.00005

So although this change improves the auto-estimate, it still doesn't work at least for the data set I've tested, probably because there are some huge artifacts within the 10s segments.

I don't know if it's worth adding this feature then. I guess as long as it doesn't make things worse it'll probably be OK to add it.

larsoner, 2019-06-05:
> So although this change improves the auto-estimate, it still doesn't work at least for the data set I've tested, probably because there are some huge artifacts within the 10s segments.

I'm fine with also changing it to use a robust (non-parametric) variance estimator instead of `std` (assuming that's what we use).

> I'm still interested on how you handle such cases though.

You just nest the import in the function that needs it:
```
def _whatever_scaling(params):
    from ..events import make_fixed_length_events
    ...
```

cbrnr, 2019-06-05:
> I'm fine with also changing it to use a robust (non-parametric) variance estimator instead of std (assuming that's what we use).

I think we're using the max of the 5% and 95% percentiles if I'm not mistaken. We could maybe lower this value and use something closer to the IQR.

larsoner, 2019-06-05:
Or maybe use the median absolute deviation?

cbrnr, 2019-06-05:
I'll play around with some values and see what works best in my case.

larsoner, 2019-06-05:
Sounds good. As long as it works for your case with outliers, and with 'sample' (which is pretty well behaved/Gaussian) then we should be good

PR #6413 (2019-06-13): MRG+1: Add remove_dc argument to _compute_scalings
Fixes #6406.
closed 2019-06-13 by agramfort