mne-tools/mne-python #6411: VIZ: Add a plot method to the digitization class

GuillaumeFavelier, 2019-06-05:
As discussed with @massich in [#6369(comment)](https://github.com/mne-tools/mne-python/pull/6369#issuecomment-498357971), we could review the visualization code related to digitization and add a `plot()` method to the new `Digitization` class.

agramfort, 2019-06-05:
maybe dig.plot() could just reuse the plot_alignment function?

>

GuillaumeFavelier, 2019-06-05:
Exactly what I had in mind!

massich, 2019-06-05:
my idea was to do a **first_time_issue** out of it. But never mind, I was wrong. Among other things, there are circular dependencies that make it not suited for a newcomer. @GuillaumeFavelier you can take it, or tell me how you want the visualization part and I can push in your PR, etc.

GuillaumeFavelier, 2019-06-05:
My bad, I didn't know we had such a tag @massich . I will take it as I already started

closed 2019-12-28 by agramfort