mne-tools/mne-python #6419: Repeated measures anova for unbalanced design with permutations test

AmirDJV, 2019-06-05:
I'm following the [Repeated measures ANOVA on source data with spatio-temporal clustering ](https://martinos.org/mne/stable/auto_tutorials/stats-source-space/plot_stats_cluster_spatio_temporal_repeated_measures_anova.html#sphx-glr-auto-tutorials-stats-source-space-plot-stats-cluster-spatio-temporal-repeated-measures-anova-py) tutorial with my own data. 

The solution to calculate permutations test for between  (visual vs auditory) - within (left vs right) repeated measures anova was to use the `stat_fun` argument with the [mne.stats.f_mway_rm](https://martinos.org/mne/stable/generated/mne.stats.f_mway_rm.html) function.  

My issue is that [mne.stats.f_mway_rm](https://martinos.org/mne/stable/generated/mne.stats.f_mway_rm.html) function only support fully balanced designs. Is there a way to support unbalanced designs? 

Thank you! :)

jona-sassenhagen, 2019-06-07:
@dengemann

dengemann, 2019-06-07:
No this is unfortunately not easy to do. The current implementation does not support mixed within-between models.

AmirDJV, 2019-06-08:
Thank you :) Are you thinking about implementing it? and can you please explain why it is problematic?

larsoner, 2020-03-04:
Closing as a dup of #4859

closed 2020-03-04 by larsoner