mne-tools/mne-python #6425: How do I disable the Esc key from closing raw.plot?

cbrnr, 2019-06-06:
In MNELAB, I would like my raw.plot window to not close if the user hits the Esc key. Previously, I used the following hack which worked for matplotlib < 3:

```
 try:
    key_events = fig.canvas.callbacks.callbacks["key_press_event"][8]
except KeyError:
    pass
else:  # this requires MNE >=0.15
    key_events.func.keywords["params"]["close_key"] = None
```

Of course this was bound to fail, and with matplotlib >= 3 `key_events` is now a `matplotlib.cbook._StrongRef` object, whereas previously it used to be a `matplotlib.cbook._BoundMethodProxy` object. Of course private methods can change at anytime, but does anyone know either how to make the old hack work again, or better yet how to modify/delete existing callbacks?

larsoner, 2019-06-06:
I wonder if we should just monkey patch to add an attribute to `fig` as `fig._mne_params = params`

cbrnr, 2019-06-06:
Yes, why not? If this gives an interface where I can change/delete callbacks... But once a callback is set, would adding our params solve the issue? Isn't a matplotlib call necessary to change a callback? If so, which?

larsoner, 2019-06-06:
> Isn't a matplotlib call necessary to change a callback? If so, which?

It depends what you mean. If you mean removing or adding a callback function yes you have to use mpl. If you just want to make it so that the existing `mne` callback does nothing on `esc`, it should be enough to do `fig._mne_params['close_key'] = None`.

PR #6429 (2019-06-11): Add params to raw.plot figure
Fixes #6425. This allows manipulating keybinds after the raw plot has been created, which is useful e.g. if you want to remove the Escape shortcut to close the window.
closed 2019-06-11 by agramfort