mne-tools/mne-python #6431: DOC improve description of dipole_fit procedure

tomMoral, 2019-06-07:
The [example](https://www.martinos.org/mne/stable/auto_tutorials/source-modeling/plot_dipole_fit.html#) showing how to use `dipole_fit` do not give any information on how the source localization is done.

I guess this is probably planned to land in issue #5944 but is it possible to get some resources (like the original ECD paper) to document the general idea. I might give a try if I find some time.

drammock, 2019-06-07:
@tomMoral I'm slowly working through the tutorials but I don't have a lot of experience with dipole fitting anyway so your input would be helpful.  If you open a PR I'll gladly review / help.

FYI #5944 is now sort of stale; the changes are being tackled one tutorial at a time instead of waiting until everything is finished and trying to merge them all in at once.

agramfort, 2019-06-08:
I think the question is what is a good academic reference for the dipole fitting algorithm.

maybe @mshamalainen can help here?

PR #6949 (2019-10-11): DOC add reference for dipole fit
closes #6431
closed 2019-10-11 by larsoner