mne-tools/mne-python #6444: Simpler resample to get around out of memory errors

bdyetton, 2019-06-11:
#### Describe the problem
I'm loading some very large 64 channel overnight sleep .eeg files with sizes of up to 5GB.
The sample rate is 500 for some of these files. Id like to resample to make them smaller and easier to process, however, resample crashes:

ValueError: Internal error occurred: b'Intel MKL DFTI ERROR: Not enough memory to allocate'

#### Describe your solution
It appears that resample is clever and tries to faithfully interpolate to remove any aliasing effects.
This cleverness requires too much memory on my computer. One solution might be to have a simpler version of resample which just drops every nth sample or the like. 

#### Describe possible alternatives
Happy to take suggestions :)

#### Additional context
I'm already using memory mapped files.

agramfort, 2019-06-11:
see this old issue: https://github.com/mne-tools/mne-python/issues/2792
we don't have a raw.decimate but when you have Epochs you can pass a decim parameter.

what do you need to do with Raw object? filter it?

bdyetton, 2019-06-11:
Hi @agramfort,
Yes, basically am using MNE to do a series of pre-processing steps to take our data from raw format to a sleep scoring format. This involves filtering, re-reference, dropping channels, ordering channels, etc, and then saving the RawArray as an EDF (courtesy of [skjerns](https://gist.github.com/skjerns/bc660ef59dca0dbd53f00ed38c42f6be)). 

btw, big ups on that Reinmann Geometry tutorial paper, I very much look forward to trying that in the future :)

agramfort, 2019-06-12:
ok

so you can read the data with preload=True but you cannot have more than 1
copy of it in memory?
did you consider passing a filename to preload to use a memmap file?

>

bdyetton, 2019-06-12:
Yea it was already a mem-mapped file. 
I've moved to a different computer to get around this. 
You're more than welcome to close this issue, however, +1 for RawArray.decimate in the future :), especially if this can happen preload

closed 2019-06-13 by agramfort