mne-tools/mne-python #6445: Why does MNE use adaptive z-scoring in find_bads_eog?

matthewhang, 2019-06-11:
According to the documentation of find_bads_eog
> Detection is based on Pearson correlation between the filtered data and the filtered EOG channel. Thresholding is based on adaptive z-scoring.

Are there any experiments/papers demonstrating the benefits of this? or is it just a traditional practice? (which i believe it isnt as there are many papers still using visual inspection), or is it just a practical recommendation without detailed evidence?

Thank you very much. I am concerning if such computation would be too conservative.

dengemann, 2019-06-11:
It’s original MNE cuisine :) something that turned out to work. We implemented it based on our own intuition and tested it on multiple datasets. We could add other options or change the default if we can convince ourselves of an alternative.

matthewhang, 2019-06-13:
> It’s original MNE cuisine :) something that turned out to work. We implemented it based on our own intuition and tested it on multiple datasets. We could add other options or change the default if we can convince ourselves of an alternative.
> […](#)

thank you for the answer! I was just testing around the options!

closed 2019-06-13 by matthewhang