mne-tools/mne-python #6454: BUG: ecog re-reference applied to non-data channels

kdoelling1919, 2019-06-12:
I'm trying my hand applying mne to some ecog data and noticed a weird-ish bug using `.set_eeg_reference` which perhaps I had no business using to begin with. I set a custom reference using 
```
good_inds = mne.pick_types(raw.info, ecog=True)
goodchs = [raw.ch_names[g] for g in good_inds]
raw.set_eeg_reference(ref_channels=goodchs)
```
I noticed after doing this that while the reference was successful, my trigger channel had some weird changes to it which suggests that the reference is being applied to non-data channels as well.

This seems to happen in the [_apply_reference](https://github.com/mne-tools/mne-python/blob/569924b3ddcca4245725930ebca243ef213b31ee/mne/io/reference.py#L98) function which searches for eeg channels to apply the reference to and comes up with an empty list. Then in L136 the pick_channels takes the empty list and returns ALL channels instead of none of them. 

Moral of the story: I expected the reference to apply only to the ecog channels but instead it was applied to all channels. I can see a couple of solutions to this. 
1. An error at the beginning of set_eeg_reference if there are no EEG channels. Is there an ECoG reference function I'm unaware of? Should I just be doing this by hand?
2. A more general function `set_reference()` which handles EEG as well as ECoG data more flexibly. Perhaps defaulting to common average reference for ecog grids and bipolar references for seeg electrodes as suggested in #5081

larsoner, 2019-06-12:
I really think we should get rid of the `pick_channels(ch_names, []) == ch_names` behavior. We keep hitting bugs like this, and there easily could be other bugs hiding from us.

I can work on doing this in the code base if @agramfort agrees. I guess the thing to do for now would be to raise an error if the second argument is empty. Eventually we could consider returning an empty list but maybe we won't need to. But either way, raising an error should be safer than the counterintuitive behavior we have now.

larsoner, 2019-06-12:
> A more general function set_reference() which handles EEG as well as ECoG data more flexibly. 

We should probably just make `set_eeg_reference` mean "set eeg/seeg/ecog reference" even though it's a bit of a misnomer. It's probably not a totally deliberate design choice to currently have only EEG and exclude sEEG/ECoG, but more an issue that we haven't had so many sEEG / ECoG users with some of these functiosn.

agramfort, 2019-06-13:
that's definitely a bug in set_eeg_reference

it should not look for eeg channels when ref_channels are present

that's the minimal fix to do. I would go this route.

>

larsoner, 2019-06-20:
> it should not look for eeg channels when ref_channels are present

I'm actually not sure what you mean by this. I'll open a PR with a minimal fix to avoid the error for sEEG/ECoG people, but it means that we only have code for EEG users to (re)reference their data, not ECoG or sEEG people.

PR #6478 (2019-06-22): MRG, BUG: Avoid referencing all data
Closes #6454.

@kdoelling1919 can you confirm this at least prevents the dangerous behavior you observed?
closed 2019-06-22 by agramfort