mne-tools/mne-python #6458: MNE.Picks Selection

jgeller112, 2019-06-15:
Hello, 

I am trying to select a only a few sensors for analysis and then apply AutoReject: 

``` 
ica_epochs = mne.read_epochs(ica_apply_path +subject+ 'ar'+'-epo.fif')
picks = mne.pick_types(ica_epochs.info, meg=False, eeg=True, stim=False, eog=False,
              selection=['A11', 'A3', 'A4', 'B14', 'B15', 'B4', 'B5', 'B6', 'B7'])
 ar = AutoReject(picks=picks)
```
Instead of getting the selection of sensors I requested, I get all sensors back:

```
<Info | 19 non-empty fields
    bads : list | 0 items
    ch_names : list | A1, A2, A3, A4, A5, A6, A7, A8, A9, ...
    chs : list | 64 items (EEG: 64)
    comps : list | 0 items
    custom_ref_applied : bool | True
    dev_head_t : Transform | 3 items
    dig : list | 67 items (3 Cardinal, 64 EEG)
    events : list | 0 items
    file_id : dict | 4 items
    highpass : float | 1.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 100.0 Hz
    meas_date : tuple | 2019-03-29 15:57:19 GMT
    meas_id : dict | 4 items
    nchan : int | 64
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 256.0 Hz
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    description : NoneType
    dev_ctf_t : NoneType
    experimenter : NoneType
    gantry_angle : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    proj_id : NoneType
    proj_name : NoneType
    subject_info : NoneType
    xplotter_layout : NoneType
```
Any idea why this might be?

agramfort, 2019-06-16:
can you provide a full code snippet? what is this info you show?
is it picks that contains all sensors? did you try:

picks = mne.pick_types(ica_epochs.info, meg=False, eeg=False,
stim=False, eog=False,
              selection=['A11', 'A3', 'A4', 'B14', 'B15', 'B4', 'B5',
'B6', 'B7'])

ie only have a selection?

jgeller112, 2019-06-16:
I am using AR after performing ICA and removing bad components. I am trying to look at 9 front-central electrodes. Here is full function: 

```
def autoreject_epochs(subject,ica_apply_path, clean_path, overwrite):
    ica_epochs = mne.read_epochs(ica_apply_path +subject+ 'ar'+'-epo.fif')
    #pickseeg=['A11', 'A3', 'A4', 'B14', 'B15', 'B4', 'B5', 'B6', 'B7'] # include only these channels 
    picks = mne.pick_types(ica_epochs.info, meg=False, eeg=True, stim=False, eog=False,
                 selection=['A11', 'A3', 'A4', 'B14', 'B15', 'B4', 'B5', 'B6', 'B7'])
    ar = AutoReject(ica_epochs.info, picks=picks)
    epochs_clean = ar.fit_transform(ica_epochs)
```
Sorry I was not clear in my previous post. The info provided is the result after this function is applied. Instead of the 9 electrodes I selected, I get all 64.

jgeller112, 2019-06-16:
Doing this 
```
picks = mne.pick_types(ica_epochs.info, meg=False, eeg=False,
stim=False, eog=False,
              selection=['A11', 'A3', 'A4', 'B14', 'B15', 'B4', 'B5',
'B6', 'B7'])
```
Returns this error: No channels match the selection.

agramfort, 2019-06-17:
try:

picks = mne.pick_channels(ica_epochs.ch_names, include=['A11', 'A3', 'A4',
'B14', 'B15', 'B4', 'B5', 'B6', 'B7'])

jgeller112, 2019-06-17:
Nope. 

```
picks = mne.pick_channels(d.ch_names, include=['A11', 'A3', 'A4',
'B14', 'B15', 'B4', 'B5', 'B6', 'B7'])

 array([ 2,  3, 10, 35, 36, 37, 38, 45, 46])
```
This looks right. 
```
ar = AutoReject(picks=picks)
AutoReject(consensus=array([0. , 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1. ]),
      cv=10, n_interpolate=None, n_jobs=1,
      picks=array([ 2,  3, 10, 35, 36, 37, 38, 45, 46]), random_state=None,
      thresh_method='bayesian_optimization', verbose='progressbar')
```

This looks right. 

```
<Info | 19 non-empty fields
    bads : list | A14
    ch_names : list | A1, A2, A3, A4, A5, A6, A7, A8, A9, ...
    chs : list | 64 items (EEG: 64)
    comps : list | 0 items
    custom_ref_applied : bool | True
    dev_head_t : Transform | 3 items
    dig : list | 67 items (3 Cardinal, 64 EEG)
    events : list | 0 items
    file_id : dict | 4 items
    highpass : float | 1.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 100.0 Hz
    meas_date : tuple | 2018-06-27 13:17:57 GMT
    meas_id : dict | 4 items
    nchan : int | 64
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 256.0 Hz
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    description : NoneType
    dev_ctf_t : NoneType
    experimenter : NoneType
    gantry_angle : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    proj_id : NoneType
    proj_name : NoneType
    subject_info : NoneType
    xplotter_layout : NoneType
```
This does not. 

Is there an easier way to select electrodes for analysis?  I guess I can manually exclude the channels if nothing comes to mind.

agramfort, 2019-06-17:
I am confused. What is the Info you show at the end? where is it coming
from?
what you provide picks to AutoReject you limit cleanup to these channels
and the
rest are left untouched I think.

>

jgeller112, 2019-06-17:
The info is after the AutoReject. 

Hmm. Is there an easy way to limit the electrodes used for epoched or evoked objects? I want to perform a cluster test on only those 9 electrodes.

drammock, 2019-06-17:
you could reduce the epochs in advance like this
```python
epochs_picked = epochs.copy().pick_channels([list_of_channel_names])
```
and then use `epochs_picked` thereafter

jgeller112, 2019-06-17:
Oh! Thank you!

closed 2019-06-18 by agramfort

jasmainak, 2019-06-18:
yes autoreject only limits analysis to the "picks" channels. It does not subselect the epochs object