mne-tools/mne-python #6461: Digitization progress

massich, 2019-06-17:
This issue is to keep track of the remaining Digitization PRs. It reflects a conversation with @larsoner.

TODO:

1- Add `montage` to all readers (aka `BaseRaw.__init__(..., montage=montage)` and calling `self.set_montage(montage)`)
2- Add missing readers to `_test_raw_reader`
3- ensure `None` or `Digitization` in `_test_raw_reader`
4- deprecate all montage related arguments but `montage` (for all readers and make sure that they are set through `BaseRaw` and `Digitization`

Here is a summary of the readers with respect to how they are tested. Main focus: does the reader go through `_test_raw_reader`? has the reader been tested with `digitization` or `None`? Does the reader have digitization or a `montage`?

```sh
                       | _test_raw_reader | None | Digitization | other   |
-----------------------+------------------+------+--------------+---------+
RawArray               |     x            |  x   |              |         |
read_raw_artemis123    |     x            |      |     x        |         |   
read_raw_bdf           |                  |      |              | montage |
read_raw_brainvision   |     x            |  x   |     x        |         |    
read_raw_bti           |     x            |  x   |     x        |         |
read_raw_cnt           |     x            |  x   |              | montage |
read_raw_ctf           |     x            |      |     x        |         |
read_raw_edf           |     x            |  x   |     x        |         |
read_raw_eeglab        |     x            |      |     x        |         |
read_raw_egi           |     x            |  x   |              | montage |
read_raw_eximia        |     x            |  x   |              | ------- |
read_raw_fieldtrip     |                  |      |              | ------- |
read_raw_fif           |     x            |      |     x        |         |
read_raw_gdf           |     x            |  x   |              | montage |
read_raw_kit           |     x            |  x   |     x        |         |
read_raw_nicolet       |     x            |  x   |              | montage |
```

Log of past PRs:
* MRG: Prepare BTI to refactor for Digitization (#6228)
* MAINT: Refactor digitation for KIT, artemis123, bti (#6368)
* ENH: Add digitization class (#6369)

Cross-reference issues:

GuillaumeFavelier, 2019-06-17:
Thanks for the overview @massich! I think there is also #6412 (add a `plot()` function to the `Digitization` class). I still have to integrate @agramfort reviews.

massich, 2019-06-17:
@GuillaumeFavelier right!

larsoner, 2019-06-18:
@massich updated top description with my suggested order / description based on extrapolating what I discussed with @agramfort

massich, 2019-06-19:
just to make sure that we are all on the same page. This should pass without warning error nothing:

```py
def test_montage():
    raw_montage = read_raw_*(fname, montage=my_montage)
    raw_none = read_raw_*(fname, montage=None)
    raw_none.set_montage(montage)
    assert object_diff(raw_none.info['chs'], raw_montage.info['chs'])
    assert object_diff(raw_none.info['dig'], raw_montage.info['dig'])
```

agramfort, 2019-06-19:
yes exactly

>

larsoner, 2019-06-19:
Yes in this PR. Eventually in this release cycle we will deprecade the `read_raw_*(fname, montage=my_montage)`, but not in this PR. (It should be almost trivial after this PR.)

larsoner, 2019-06-19:
Sorry, I meant to say that yes, that should be the behavior in #6462.

In the PR after that, `read_raw_*(fname, montage=my_montage)` should be deprecated and the user told to do `raw.set_montage(my_montage)`.

massich, 2019-06-26:
`update_ch_names` has been exposed in 0.19. Once Digitization is here, we should be able to remove it from the signature. 
https://github.com/mne-tools/mne-python/blob/2f9873e7759e62f8471326e37a2ad1cdd2c2e65d/mne/channels/channels.py#L446

In other words, `Digitization` should go in before releasing so that we don't need to deprecate this parameter.

massich, 2019-07-10:
now that montage is deprecated, the idea is that `raw.set_montage` only takes Digitization and that as a middle step montages and digmontages had to be transformed to Digitizations, am I right? @agramfort, @larsoner

larsoner, 2019-07-10:
Yeah I think so.

agramfort, 2019-07-10:
question is should we still call it set_montage or we introduce a new
method set_digitization?
just thinking out loud...

massich, 2019-07-11:
`set_montage` sounds fine to me

massich, 2019-07-11:
There are at least two ways of getting the digitization information from a fif file:

https://github.com/mne-tools/mne-python/blob/897c3b2d12d0562f888784c5e18c2da31ed27ceb/mne/io/meas_info.py#L891

https://github.com/mne-tools/mne-python/blob/897c3b2d12d0562f888784c5e18c2da31ed27ceb/mne/channels/montage.py#L637

agramfort, 2019-07-11:
not really as the anything is done after _read_dig_fif(fid, tree) is
just to make a DigMontage from a dig (get numpy arrays and a few attributes)

>

massich, 2019-08-09:
I think that we could simplify mne/viz/montage.py quite a lot. see #6649

massich, 2019-08-15:
cross-reference https://github.com/mne-tools/mne-python/issues/3893#issuecomment-271925975 and #6461

massich, 2019-08-26:
Regarding the `kind` we have this `FIF.FIFV_MNE_COORD_DIGITIZER` and `FIFFV_COORD_ISOTRAK` that is used nowhere.

```sh
~/code/mne-python read_dig_montage_polhemus
(mne) ❯ git grep -i 'FIFFV_MNE_COORD_DIGITIZER'
mne/io/constants.py:FIFF.FIFFV_MNE_COORD_DIGITIZER   = FIFF.FIFFV_COORD_ISOTRAK # Original (Polhemus) digitizer coordinates
mne/io/tests/test_constants.py:    FIFFV_MNE_COORD_DIGITIZER='FIFFV_COORD_ISOTRAK',
```

agramfort, 2019-08-26:
yes then we need to change the read_dig_xxx so they don't use unknown but
the proper
source of the data. We would need to add a constant for captrack though

larsoner, 2019-09-21:
@massich do we need any of the rest of these for 0.19? Or just #6764?

massich, 2019-09-23:
I think #6764 is enough. I'm closing this one

closed 2019-09-23 by massich