mne-tools/mne-python #6463: MEGSIM is unrealiable

massich, 2019-06-17:
http://cobre.mrn.org/megsim/ keeps breaking our CricleCI. We mock it for the testing in #6453 but our CircleCI in master is completely random due to the fetching.

Could we mirror it?

jhouck, 2019-06-17:
Briefly, the issue is that MRN thought no one was using it, and it was hosted externally, so they took it down to save on the hosting costs.  Ideally they'll resurrect it at the same address, since the published manuscript also points to that URL.  That said, if you have an easy way to mirror it, that might still be a good idea. 

I'm affiliated with MRN, not just randomly butting in. :)

agramfort, 2019-06-18:
we could put it on osf.io

>

larsoner, 2019-06-19:
> MRN thought no one was using it

If this is the case, maybe they are right. Perhaps we should just remove the dataset, especially given the progress we have made on making the simulation module more usable.

agramfort, 2019-06-19:
no strong feeling here. It's true that little people have actually used
these data in the last years

>

larsoner, 2019-06-19:
I'll make a PR to remove it. We are early in the release cycle so if it gets reuploaded somewhere we can easily revert the commit

PR #6472 (2019-06-19): MRG, MAINT: Remove megsim dataset
Closes #6463.
closed 2019-06-19 by larsoner