mne-tools/mne-python #6467: BUG: Import raw CNT file channels in integer format; not strings

biancaisla1, 2019-06-18:
Having trouble importing raw .CNT file. Previous steps include importing os, importing MNE, importing numpy as np, importing mayavi, setting log level, importing cnt, defining the file path, and defining the raw_fname.  When trying to use read function I get the following:
```
In [12]: raw = mne.io.read_raw_cnt(input_fname='./raw/s1066.cnt', montage=('montage'), eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'), 
                          misc=('SCR', 'SCL'), ecg=('HR'), 
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'), 
                          data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-12-a085b95a7d93>:4: RuntimeWarning:   Could not parse meas date from the header. Setting to None.
  data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-12-a085b95a7d93>:4: RuntimeWarning: Could not define the number of bytes automatically. Defaulting to 2.
  data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-12-a085b95a7d93> in <module>
      2                           misc=('SCR', 'SCL'), ecg=('HR'),
      3                           emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'),
----> 4                           data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in read_raw_cnt(input_fname, montage, eog, misc, ecg, emg, data_format, date_format, preload, stim_channel, verbose)
    176     return RawCNT(input_fname, montage=montage, eog=eog, misc=misc, ecg=ecg,
    177                   emg=emg, data_format=data_format, date_format=date_format,
--> 178                   preload=preload, stim_channel=stim_channel, verbose=verbose)
    179 
    180 

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in __init__(self, input_fname, montage, eog, misc, ecg, emg, data_format, date_format, preload, stim_channel, verbose)
    430         input_fname = path.abspath(input_fname)
    431         info, cnt_info = _get_cnt_info(input_fname, eog, ecg, emg, misc,
--> 432                                        data_format, _date_format, stim_channel)
    433         last_samps = [cnt_info['n_samples'] - 1]
    434         _check_update_montage(info, montage)

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in _get_cnt_info(input_fname, eog, ecg, emg, misc, data_format, date_format, stim_channel_toggle)
    317 
    318     chs = _create_chs(ch_names, cals, FIFF.FIFFV_COIL_EEG,
--> 319                       FIFF.FIFFV_EEG_CH, eog, ecg, emg, misc)
    320     eegs = [idx for idx, ch in enumerate(chs) if
    321             ch['coil_type'] == FIFF.FIFFV_COIL_EEG]

~\Anaconda3\envs\mne\lib\site-packages\mne\io\utils.py in _create_chs(ch_names, cals, ch_coil, ch_kind, eog, ecg, emg, misc)
    278             coil_type = FIFF.FIFFV_COIL_NONE
    279             kind = FIFF.FIFFV_EOG_CH
--> 280         elif ch_name in ecg or idx in ecg:
    281             coil_type = FIFF.FIFFV_COIL_NONE
    282             kind = FIFF.FIFFV_ECG_CH

TypeError: 'in <string>' requires string as left operand, not int/
I found this reference that talks about how a Neuroscan EEG file is formatted, may be it can be of some use?
http://paulbourke.net/dataformats/eeg/

agramfort, 2019-06-18:
can you share the problematic file privately?

biancaisla1, 2019-06-18:
Good Evening Alexandre-Gramfort,

I  am a research assistant in Dr. Stephen Benning’s lab at UNLV.  I am doing the groundwork to transition our lab to MNE-Python, but I believe Dr. Benning would make the decision about sharing the CNT file. Of course this decision would be based on whether the lab wants to publish the data from the file at any point in the future.  I have cc’d my boss here, and hope to see his response by the morning (PST).  If this is the case and this particular script is private, then maybe we can come up with a faux script that will suffice until the issue can be resolved. 

Best,
Bianca Islas

From: Alexandre Gramfort
Sent: Tuesday, June 18, 2019 1:06 AM
To: mne-tools/mne-python
Cc: biancaisla1; Manual
Subject: Re: [mne-tools/mne-python] BUG: Import raw CNT file channels ininteger format; not strings (#6467)

can you share the problematic file privately?

massich, 2019-06-18:
The following replicates the same error:

```sh
In [1]: 3 in ('hi')                                                                                     
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-1-4c561b2f4433> in <module>
----> 1 3 in ('hi')

TypeError: 'in <string>' requires string as left operand, not int
```

I've looked at your trace it should be good to know whats the value of `ch_names` To do so you should just re-run the same and after the failure you run `debug` the prompt will change to something that would look like `ipdb>` then you need to go one function upwards (by typing `u`) then you can print the value `print(ch_names)`. I'll try to mock up what you should be seeing:

```sh

In [12]: raw = mne.io.read_raw_cnt(input_fname='./raw/s1066.cnt', montage=('montage'), eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'), 
                          misc=('SCR', 'SCL'), ecg=('HR'), 
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'), 
                          data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-12-a085b95a7d93>:4: RuntimeWarning:   Could not parse meas date from the header. Setting to None.
  data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-12-a085b95a7d93>:4: RuntimeWarning: Could not define the number of bytes automatically. Defaulting to 2.
  data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-12-a085b95a7d93> in <module>
      2                           misc=('SCR', 'SCL'), ecg=('HR'),
      3                           emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'),
----> 4                           data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in read_raw_cnt(input_fname, montage, eog, misc, ecg, emg, data_format, date_format, preload, stim_channel, verbose)
    176     return RawCNT(input_fname, montage=montage, eog=eog, misc=misc, ecg=ecg,
    177                   emg=emg, data_format=data_format, date_format=date_format,
--> 178                   preload=preload, stim_channel=stim_channel, verbose=verbose)
    179 
    180 

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in __init__(self, input_fname, montage, eog, misc, ecg, emg, data_format, date_format, preload, stim_channel, verbose)
    430         input_fname = path.abspath(input_fname)
    431         info, cnt_info = _get_cnt_info(input_fname, eog, ecg, emg, misc,
--> 432                                        data_format, _date_format, stim_channel)
    433         last_samps = [cnt_info['n_samples'] - 1]
    434         _check_update_montage(info, montage)

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in _get_cnt_info(input_fname, eog, ecg, emg, misc, data_format, date_format, stim_channel_toggle)
    317 
    318     chs = _create_chs(ch_names, cals, FIFF.FIFFV_COIL_EEG,
--> 319                       FIFF.FIFFV_EEG_CH, eog, ecg, emg, misc)
    320     eegs = [idx for idx, ch in enumerate(chs) if
    321             ch['coil_type'] == FIFF.FIFFV_COIL_EEG]

~\Anaconda3\envs\mne\lib\site-packages\mne\io\utils.py in _create_chs(ch_names, cals, ch_coil, ch_kind, eog, ecg, emg, misc)
    278             coil_type = FIFF.FIFFV_COIL_NONE
    279             kind = FIFF.FIFFV_EOG_CH
--> 280         elif ch_name in ecg or idx in ecg:
    281             coil_type = FIFF.FIFFV_COIL_NONE
    282             kind = FIFF.FIFFV_ECG_CH

TypeError: 'in <string>' requires string as left operand, not int```

In [13]: debug

~\Anaconda3\envs\mne\lib\site-packages\mne\io\utils.py in _create_chs(ch_names, cals, ch_coil, ch_kind, eog, ecg, emg, misc)
    278             coil_type = FIFF.FIFFV_COIL_NONE
    279             kind = FIFF.FIFFV_EOG_CH
--> 280         elif ch_name in ecg or idx in ecg:
    281             coil_type = FIFF.FIFFV_COIL_NONE
    282             kind = FIFF.FIFFV_ECG_CH

ipdb> u

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in _get_cnt_info(input_fname, eog, ecg, emg, misc, data_format, date_format, stim_channel_toggle)
    317 
    318     chs = _create_chs(ch_names, cals, FIFF.FIFFV_COIL_EEG,
--> 319                       FIFF.FIFFV_EEG_CH, eog, ecg, emg, misc)
    320     eegs = [idx for idx, ch in enumerate(chs) if
    321             ch['coil_type'] == FIFF.FIFFV_COIL_EEG]

ipdb> print(ch_names)                                                   
['1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23', '24', '25', '26', '27', '28', 'LEFT_EAR', 'VEOGR', '121', '122', '29', '30', '31', '32', '33', '34', '35', '36', '37', '38', '39', '40', '41', '42', '43', '44', '45', '46', '47', '48', '49', '50', '51', '52', '53', '54', '55', '56', 'HEOG', 'NA1', '123', '124', '57', '58', '59', '60', '61', '62', '63', '64', '65', '66', '67', '68', '69', '70', [... cut ...] '76', '77', '78', '79', '80', '81', '82', '83', '84', '85', '86', '87', '88', '89', '90', '91', '92', '93', '94', '95', '96', '97', '98', '99', '100', '101', '102', '103', '104', '105', '106', '107', '108', '109', '110', '111', '112', '113', '114', '115', '116', '117', '118', '119', '120']

ipdb>                                            

```

Notice that in my case `ch_names` are strings. Your error says that they are integers rather than strings which is weird. Can you check that for me? thx

biancaisla1, 2019-06-18:
> *From: *Alexandre Gramfort <notifications@github.com>
> *To: *mne-tools/mne-python <mne-python@noreply.github.com>
> *Cc: *biancaisla1 <biancaisla1@gmail.com>; Manual
> .[image:

I'd be glad to share the files we've been trying to import. Having the real
files will be invaluable for this kind of debugging. I've got three sample
files, all of which have been successfully imported using a custom Matlab
toolbox and which are readable in Scan from the vendor's software.

How would you like us to send them to you?

-- 
Stephen D. Benning
Assistant professor
Department of Psychology
4505 S. Maryland Pkwy, Box 455030
Las Vegas, NV 89154-5030
Phone: 702-895-5061
Office: CBC B339, Lab: CBC B147
Website: http://www.peplab.org

*Please Note:  The information transmitted in this email is intended only
for the person or entity to which it is addressed and may contain
confidential and/or privileged material.  Any review, transmission,
dissemination, or other use of or taking of any action in reliance upon
this information by persons or entities other than the intended recipient
is prohibited.  If you receive this email in error, please contact the
sender and delete the material from any computer. *

larsoner, 2019-06-18:
Usually a dropbox/google drive link works well (or really any link that lives for ~1 mo), that way multiple devs can have a look

cbrnr, 2019-06-19:
@massich I think the question is why `ecg` is a string and not a tuple here. `idx` will always be an integer, so we do `idx in ecg`, which works if `ecg` is a tuple (or a list), but it fails if it is a string.

biancaisla1, 2019-06-19:
MNE team,

We have created a Google drive to share the CNT files.  Is mne-python@noreply.github.com a good email address to send the permissions?

Bianca Islas

From: Clemens Brunner
Sent: Wednesday, June 19, 2019 1:07 AM
To: mne-tools/mne-python
Cc: biancaisla1; Manual
Subject: Re: [mne-tools/mne-python] BUG: Import raw CNT file channels ininteger format; not strings (#6467)

@massich I think the question is why ecg is a string and not a tuple here. idx will always be an integer, so we do idx in ecg, which works if ecg is a tuple (or a list), but it fails if it is a string.

larsoner, 2019-06-19:
Can you just make a publicly shared link? It's easier than sending email addresses

biancaisla1, 2019-06-19:
https://drive.google.com/file/d/14jJK1tvfg2J4VPxZf_QBeVlVQnbV9eop/view?usp=sharing
https://drive.google.com/file/d/1VZP-V4TM0hsX3dW8WFy0XBKFZ_tOXrOD/view?usp=sharing
https://drive.google.com/file/d/1oHfDv3Ik1V_lihuXeDqrjSkFFmCyo_eW/view?usp=sharing

We are sharing three EEG CNT files, so there is 3 links above. Please let me know if they work, and if there is anyway I may further assist the team.
Bianca Islas
From: Eric Larson
Sent: Wednesday, June 19, 2019 9:51 AM
To: mne-tools/mne-python
Cc: biancaisla1; Manual
Subject: Re: [mne-tools/mne-python] BUG: Import raw CNT file channels ininteger format; not strings (#6467)

Can you just make a publicly shared link? It's easier than sending email addresses

agramfort, 2019-06-20:
this works:

```
import mne

raw = mne.io.read_raw_cnt(input_fname='st003_PAMR_s1093_20080414.cnt',
                          eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'), 
                          misc=('SCR', 'SCL'), ecg=('HR',), 
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'), 
                          data_format='auto', date_format='mm/dd/yy', preload=True)
```

you need to be careful that `('HR')` is equivalent to `'HR'` while you should put `('HR',)` or `['HR']` to provide a tuple or a list of channel names.

I had also to remove the montage parameter.

close if it's ok for you

biancaisla1, 2019-06-20:
Alexandre,

I am using Windows 10 and a Jupyter notebook (fyi) I tried to run this this morning but it is giving me a type error:

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-10-efbf9192d8ab> in <module>
      2                           misc=('SCR', 'SCL'), ecg=('HR',),
      3                           emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'),
----> 4                           data_format='auto', date_format='mm/dd/yy', preload=True)

TypeError: read_raw_cnt() missing 1 required positional argument: 'montage'

It does not however, say anything about the channel names this time so I went to print them: print(raw.ch_names)
And it gave me another error:

--------------------------------------------------------------------------- 
NameError Traceback                                 (most recent call last) 
<ipython-input-11-a4fe2c9fe6e7> in <module> 
----> 1 print(raw.ch_names) 

NameError: name 'raw' is not defined

As far as the montage goes, I have seen other people use 10-20 international standard as their montage in other examples, I believe this will suffice for our needs.  Should we try that? Also, I’m not sure why ‘raw’ is not defined?

Thank you for your patience, attention, and knowledge in this matter,
Bianca Islas
From: Alexandre Gramfort
Sent: Thursday, June 20, 2019 5:27 AM
To: mne-tools/mne-python
Cc: biancaisla1; Manual
Subject: Re: [mne-tools/mne-python] BUG: Import raw CNT file channels ininteger format; not strings (#6467)

this works:
import mne

raw = mne.io.read_raw_cnt(input_fname='st003_PAMR_s1093_20080414.cnt',
                          eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'), 
                          misc=('SCR', 'SCL'), ecg=('HR',), 
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'), 
                          data_format='auto', date_format='mm/dd/yy', preload=True)
you need to be careful that ('HR') is equivalent to 'HR' while you should put ('HR',) or ['HR'] to provide a tuple or a list of channel names.
I had also to remove the montage parameter.
close if it's ok for you

massich, 2019-06-20:
you should call it with `montage=None` at least for now. That should work.

massich, 2019-06-20:
Meanwhile, can I see which was the exact call you typed? because `montage` should not be a positional argument. It's an extra parameter that defaults to `None`.

biancaisla1, 2019-06-20:
raw_file = mne.io.read_raw_cnt(input_fname, montage, eog=(‘VEO’, ‘HEO’), misc=(‘SCR’, ‘SCL’), ecg=(‘HR’), emg=(‘PA-L-SCALP’,’PA-L-EAR’, ‘PA-R-SCALP’,’PA-R-EAR’, ‘ORB-NOSE’,’ORB-EAR’, ‘COR-NOSE’,’COR-EAR’, ‘ZYG-MOUTH’,’ZYG-EAR’), data_format='auto', date_format='mm/dd/yy', preload=True, verbose=True)

The above was the original input, now that I’m looking at it I can see what you’re saying. It was in there, it just wasn’t exactly doing/defining anything. I’ve edited this issue, but now the issue has become:

In [8]: from mne.io import cnt
           data_path = ‘C:/Users/bianc’
	raw_fname = data_path + ‘raw/st003_PAMR_1093_20080414.cnt’
	print(raw_fname)
Out [8]: C:/Users/bianc/raw/st003_PAMR_1093_20080414.cnt

In [9]: raw = mne.io.read_raw_cnt(input_fname=’st003_PAMR_s1093_20080414.cnt’, montage=None, eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'), 
                          misc=('SCR', 'SCL'), ecg=('HR',), 
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'), 
                          data_format='auto', date_format='mm/dd/yy', preload=True)
Out [9]: <ipython-input-14-428665bccef2>:4: FutureWarning: The parameter `stim_channel` controlling the stim channel synthesis has not been specified. In 0.18 it defaults to True but will change to False in 0.19 (when no stim channel synthesis will be allowed) and be removed in 0.20; migrate code to use `stim_channel=False` and :func:`mne.events_from_annotations` or set `stim_channel=True` to avoid this warning.
  data_format='auto', date_format='mm/dd/yy', preload=True)

---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-14-428665bccef2> in <module>
      2                           misc=('SCR', 'SCL'), ecg=('HR',),
      3                           emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'),
----> 4                           data_format='auto', date_format='mm/dd/yy', preload=True)

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in read_raw_cnt(input_fname, montage, eog, misc, ecg, emg, data_format, date_format, preload, stim_channel, verbose)
    176     return RawCNT(input_fname, montage=montage, eog=eog, misc=misc, ecg=ecg,
    177                   emg=emg, data_format=data_format, date_format=date_format,
--> 178                   preload=preload, stim_channel=stim_channel, verbose=verbose)
    179 
    180 

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in __init__(self, input_fname, montage, eog, misc, ecg, emg, data_format, date_format, preload, stim_channel, verbose)
    430         input_fname = path.abspath(input_fname)
    431         info, cnt_info = _get_cnt_info(input_fname, eog, ecg, emg, misc,
--> 432                                        data_format, _date_format, stim_channel)
    433         last_samps = [cnt_info['n_samples'] - 1]
    434         _check_update_montage(info, montage)

~\Anaconda3\envs\mne\lib\site-packages\mne\io\cnt\cnt.py in _get_cnt_info(input_fname, eog, ecg, emg, misc, data_format, date_format, stim_channel_toggle)
    189     # Reading only the fields of interest. Structure of the whole header at
    190     # http://paulbourke.net/dataformats/eeg/
--> 191     with open(input_fname, 'rb', buffering=0) as fid:
    192         fid.seek(21)
    193         patient_id = read_str(fid, 20)

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\bianc\\st003_PAMR_s1093_20080414.cnt'

Now I’m completely lost!

From: Joan Massich
Sent: Thursday, June 20, 2019 10:05 AM
To: mne-tools/mne-python
Cc: biancaisla1; Manual
Subject: Re: [mne-tools/mne-python] BUG: Import raw CNT file channels ininteger format; not strings (#6467)

Meanwhile, can I see which was the exact call you typed? because montage should not be a positional argument. It's an extra parameter that defaults to None.

drammock, 2019-06-20:
It looks like you are defining the full file path as `raw_fname` but then not using it within the `read_raw_cnt` function. Try:

```python
import mne
raw = mne.io.read_raw_cnt(input_fname='the/full/path/to/your/file/not/just/the/filename.cnt',
                          montage=None,
                          eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'),
                          misc=('SCR', 'SCL'),
                          ecg=('HR',),
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE',
                          	   'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'),
                          data_format='auto',
                          date_format='mm/dd/yy',
                          preload=True)
```
where `'the/full/path/to/your/file/not/just/the/filename.cnt'` is presumably `'C:/Users/bianc/raw/st003_PAMR_1093_20080414.cnt'` (based on your previous message)

biancaisla1, 2019-06-21:
My apologies for the delay in response, I plugged the corrections in this morning and this was the output I got:  

In [15]: raw = mne.io.read_raw_cnt(input_fname='C:/Users/bianc/raw/st003_PAMR_s1093_20080414.cnt', montage=None, eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'), 
                          misc=('SCR', 'SCL'), ecg=('HR',), 
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE', 'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'), 
                          data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
Out [15]: <ipython-input-15-065d7aeeb396>:4: RuntimeWarning:   Could not parse meas date from the header. Setting to None.
  		data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-15-065d7aeeb396>:4: RuntimeWarning: Could not define the number of bytes automatically. Defaulting to 2.
  		data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-15-065d7aeeb396>:4: RuntimeWarning:   Could not parse meas date from the header. Setting to None.
  		data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)
<ipython-input-15-065d7aeeb396>:4: RuntimeWarning: Could not define the number of bytes automatically. Defaulting to 2.
  		data_format='auto', date_format='mm/dd/yy', preload=True, stim_channel=True)

Also, it would appear that printing the channel names, DOES work now.

In [16]: print(raw.ch_names)
Out [16]: ['FP1', 'FPZ', 'FP2', 'F7', 'F3', 'FZ', 'F4', 'F8', 'PA-L-SCAL', 'PA-L-EAR', 'PA-R-SCAL', 'PA-R-EAR', 'C3', 'CZ', 'C4', 'M1', 'ORB-NOSE', 'ORB-EAR', 'P3', 'PZ', 'P4', 'P8', 'ZYG-MOUTH', 'ZYG-EAR', 'COR-NOSE', 'COR-EAR', 'O1', 'OZ', 'O2', 'M2', 'T8', 'T7', 'P7', 'HEO', 'VEO', 'HR', 'SCR', 'SCL', 'STI 014']
['FP1', 'FPZ', 'FP2', 'F7', 'F3', 'FZ', 'F4', 'F8', 'PA-L-SCAL', 'PA-L-EAR', 'PA-R-SCAL', 'PA-R-EAR', 'C3', 'CZ', 'C4', 'M1', 'ORB-NOSE', 'ORB-EAR', 'P3', 'PZ', 'P4', 'P8', 'ZYG-MOUTH', 'ZYG-EAR', 'COR-NOSE', 'COR-EAR', 'O1', 'OZ', 'O2', 'M2', 'T8', 'T7', 'P7', 'HEO', 'VEO', 'HR', 'SCR', 'SCL', 'STI 014']

From: Daniel McCloy
Sent: Thursday, June 20, 2019 11:55 AM
To: mne-tools/mne-python
Cc: biancaisla1; Manual
Subject: Re: [mne-tools/mne-python] BUG: Import raw CNT file channels ininteger format; not strings (#6467)

It looks like you are defining the full file path as raw_fname but then not using it within the read_raw_cnt function. Try:
import mne
raw = mne.io.read_raw_cnt(input_fname='the/full/path/to/your/file/not/just/the/filename.cnt',
                          montage=None,
                          eog=('VEOU', 'VEOL', 'HEOR', 'HEOL'),
                          misc=('SCR', 'SCL'),
                          ecg=('HR',),
                          emg=('PA-L-SCALP', 'PA-L-EAR', 'PA-R-SCALP', 'PA-R-EAR', 'ORB-NOSE',
                          	   'ORB-EAR', 'COR-NOSE', 'COR-EAR', 'ZYG-MOUTH', 'ZYG-EAR'),
                          data_format='auto',
                          date_format='mm/dd/yy',
                          preload=True)
where 'the/full/path/to/your/file/not/just/the/filename.cnt' is presumably 'C:/Users/bianc/raw/st003_PAMR_1093_20080414.cnt' (based on your previous message)

agramfort, 2019-06-21:
Glad it works !

closed 2019-06-21 by agramfort