mne-tools/mne-python #6479: API: Deprecate PSF/CTF

larsoner, 2019-06-20:
@olafhauk recently @agramfort pointed me toward:

https://www.biorxiv.org/content/10.1101/672956v1

WDYT about deprecating MNE's PSF/CTF functionality, and instead change the existing examples to use your toolbox, with a link to your package?

If you can make a PR to update the examples to use your package, I can push commits to get the CIs (and requirements files, etc.) to pull in your package properly.

jasmainak, 2019-06-20:
I think this would be a great fit for the upcoming coding sprint that @olafhauk expressed interest in attending.

larsoner, 2019-09-11:
Looks like this will not make it to 0.19, removing milestone

agramfort, 2019-09-11:
@olafhauk will be at the MNE code sprint in Boston. It should be a good time to fix this or remove it completly

PR #6877 (2019-10-03): Resolution matrix
closes #6479
closed 2019-10-03 by agramfort