mne-tools/mne-python #6485: RFC, DOC, STY: update website fonts?

drammock, 2019-06-24:
I'm often bothered by the font mismatch between the API links and the surrounding text. Here's a screenshot:

[image]

The height difference is most jarring, but there are also edge-spacing issues, so that a line that starts with an API link will look slightly indented.  The proposal here is to switch to using [Source Code Pro](https://github.com/adobe-fonts/source-code-pro) for API links and code blocks, and [Source Sans Pro](https://github.com/adobe-fonts/source-sans-pro) for the main text. Here's a mock-up:

[image]

The [Source Code](https://adobe-fonts.github.io/source-code-pro/) and [Source Sans](https://adobe-fonts.github.io/source-sans-pro/) fonts are both available as WOFF or WOFF2 format, come in several weights, and the monospace has an italic version, so there's some flexibility in fine-tuning the weights (e.g., could use medium or semibold instead of bold for the API links if we wanted to).

agramfort, 2019-06-24:
no objection

>

larsoner, 2019-06-24:
Sounds good

PR #6499 (2019-06-27): DOC, STY: Change fonts for documentation website
Closes #6485 

- uses Source Code Pro and Source Sans Pro as default fonts on website
- adjusts font size, weight, and padding for inline `code` / `pre` text
- adds font files in EOT, WOFF, OTF, TTF formats (note: only includes font weights in use, omits ExtraLight, Light, Medium, and Black)
closed 2019-06-27 by larsoner