mne-tools/mne-python #6498: BUG: Arrow rendering no longer correct (scale issue?)

larsoner, 2019-06-26:
In 0.16 it looks good

https://mne-tools.github.io/0.16/auto_tutorials/plot_dipole_orientations.html#limiting-orientations-but-not-fixing-them

In 0.17+ the activity disappears for some reason (maybe meters vs mm in the arrow scales?)

https://13931-1301584-gh.circle-artifacts.com/0/dev/auto_tutorials/source-modeling/plot_dipole_orientations.html#limiting-orientations-but-not-fixing-them

larsoner, 2019-06-26:
@GuillaumeFavelier can you look into why it no longer plots?

GuillaumeFavelier, 2019-06-26:
Yeah sure, I'll take care of it.

GuillaumeFavelier, 2019-06-28:
**TL:DR**: I removed [this line](https://github.com/mne-tools/mne-python/blob/master/mne/viz/_3d.py#L2232) and it worked for me.

For reference, this bug occurs in [`plot_dipole_orientations.py`](https://github.com/mne-tools/mne-python/blob/master/tutorials/source-modeling/plot_dipole_orientations.py#L206-L207).

I investigated this one and I discovered that the bug is related to the `alpha` value passed to the `PySurfer` `Brain` object through [`add_data()`](https://github.com/mne-tools/mne-python/blob/master/mne/viz/_3d.py#L2224-L2231).

In my experiments, I saw that although `alpha` and `vector_alpha` are passed as different parameters (`alpha` for the opacity of the overlay and `vector_alpha` for the opacity of the arrows), `alpha` has also an influence on the glyphs displayed. For short, to display the glyphs, both `alpha` and `vector_alpha` must allow it with strictly positive values.

So it's possible to do:
```diff
brain_loose = stc.plot(subjects_dir=subjects_dir, initial_time=time_max,
- time_unit='s', size=(600, 400), overlay_alpha=0)
+ time_unit='s', size=(600, 400), overlay_alpha=0.5)
```
But this will activate the smoothing steps and the rendering will be slightly different.
What do you think @larsoner ?

GuillaumeFavelier, 2019-06-28:
As a hack to disable the overlay in PySurfer, I also tried adding:

`surf.actor.visible = False`

just after [this line](https://github.com/nipy/PySurfer/blob/master/surfer/viz.py#L3272):

`surf.actor.property.backface_culling = False`

larsoner, 2019-06-28:
I'm not sure what the right thing is, I'd just make sure that the examples using vector and non-vector STC plotting still look correct (or in the case of `plot_dipole_orientations`, now look correct).