mne-tools/mne-python #6500: BUG: raw.plot_psd subplot titles broken

drammock, 2019-06-27:
in 0.18: "Magnetometers Gradiometers EEG"
https://mne-tools.github.io/stable/auto_examples/time_frequency/plot_compute_raw_data_spectrum.html#plot-the-raw-psd

in current master:  "EEG EEG EEG" and something is wrong with the confidence bands
https://mne-tools.github.io/dev/auto_examples/time_frequency/plot_compute_raw_data_spectrum.html#plot-the-raw-psd

#6451 may have broken this but I haven't dug into the blame deep enough to be sure. cc @jshanna100

jshanna100, 2019-06-27:
Ok, I know what this is.
https://github.com/mne-tools/mne-python/blob/ca870d2f04c20cee8957257df5f2d2c64d448b34/mne/viz/utils.py#L3002-L3004
The mean() operation should be removed from these, like:
`np.min(psd, axis=0)`
Then it works. I did this for 'std' but forgot to also do it for 'range'. Thanks for bringing this to my attention.

@larsoner Should I do a quick PR for this?

larsoner, 2019-06-27:
@jshanna100 that would be great

massich, 2019-06-27:
Is there an easy way to test figures? I think matplotlib does it but I'm not sure how easy / complicate is it

drammock, 2019-06-27:
@massich in my WIP refactor of `plot_compare_evokeds` I test title [like this](https://github.com/drammock/mne-python/blob/refactor-epochs-image/mne/viz/tests/test_evoked.py#L228-L231)

drammock, 2019-07-11:
this was fixed in #6506

closed 2019-07-11 by drammock

massich, 2019-07-12:
@drammock link is broken are you referring to #5944 ?

drammock, 2019-07-12:
oops, that branch was merged and deleted.  Here are the corresponding lines in master:
https://github.com/mne-tools/mne-python/blob/master/mne/viz/tests/test_evoked.py#L227-L230