mne-tools/mne-python #6502: Sanity check of co-registration quality

hichamjanati, 2019-06-27:
### problem
The trans files provided in the mne datasets are not all inspected to verify if the alignment is acceptable or not. 

### solution
We can use the following script as a test before computing a forward for example. It computes the average distance between the dig points and the mri surface (same as in the coregistration gui). 

I am not sure how to integrate this in mne but I am thinking as warning or an error when the forward is computed, depending on how large the alignment distance is. Or at least as a sanity check in the source localization examples.

```python
import warnings
import os
from mne.transforms import _get_trans, apply_trans
from sklearn.metrics import euclidean_distances

def compute_coreg_dist(subject, trans_fname, info_fname, subjects_dir):
    """Compute the avg distance between the dig points and their closest
       surface MRI neighbors in Head coordinates.
    """
    trans = mne.read_trans(trans_fname)

    high_res_surf = subjects_dir + "/%s/surf/lh.seghead" % subject
    low_res_surf = subjects_dir + "/%s/bem/%s-outer_skull.surf" % (subject,
                                                                   subject)
    if os.path.exists(high_res_surf):
        pts, _ = mne.read_surface(high_res_surf, verbose=False)
        pts /= 1e3  # convert to mm
    elif os.path.exists(low_res_surf):
        warnings.warn("""Using low resolution head surface,
                         the average distance will be potentially
                         overestimated""")
        pts, _ = mne.read_surface(high_res_surf, verbose=False)
        pts /= 1e3  # convert to mm
    else:
        raise FileNotFoundError("No MRI surface was found!")
    trans = _get_trans(trans, fro="mri", to="head")[0]
    pts = apply_trans(trans, pts)
    info = mne.io.read_info(info_fname, verbose=False)
    info_dig = np.stack([list(x["r"]) for x in info["dig"]],
                        axis=0)
    M = euclidean_distances(info_dig, pts)
    idx = np.argmin(M, axis=1)
    dist = M[np.arange(len(info_dig)), idx].mean()

    return dist
```

On the hf_sef dataset for example, we get 3-6 mm which leads to inaccurate source estimates. I will yet check the other datasets. (the authors of hf_sef already agreed to update the trans files with better ones @agramfort and I made.)

larsoner, 2019-06-27:
A `logger.info` with this information plus a warning if it's outside some range would indeed be useful. Just note that instead of `argmin(euclidean_distances(...))` we have a more efficient `_compute_nearest` that you can use.

larsoner, 2019-06-27:
Also the code from `mne.viz._3d.plot_alignment` for the `'head-dense'` argument to `'surfaces'` can be re-used for the dense head here, as can the argument for `'head'` for the low-res head surface. They are more tolerant/flexible than what is done in the current snippet.

agramfort, 2019-06-28:
we need to decide on the API and we should try to reuse as much code as
possible

thoughts anyone?

larsoner, 2019-06-28:
@jhouck do you have any code for coreg quality assessment?

agramfort, 2019-06-29:
I would suggest expose a public function like:

`check_coreg(trans, info, subject, subjects_dir=None)`

that prints the descriptive stats of the distances to haed surface (min, max, mean, 25%, 50%, 75% quantiles). Like pandas.DataFrame.describe()

ok?

larsoner, 2019-06-30:
Sounds good, and we can think about having it be called by plot_alignment if trans is not None, assuming it's fast enough

jhouck, 2019-06-30:
@larsoner yes, I threw everything into a CoregModel and used _get_point_distance. Let me find a clean version and I'll put it into a gist.

jhouck, 2019-06-30:
Ugly version at https://gist.github.com/jhouck/9585c775cb524580fc0ad2ac7f70844b .

PR #6512 (2019-07-11): MRG: check coreg function
closes #6502

still not sure about API,
no doc yet

I guess the usecase still needs to be defined. It is just to log stuff,
should this be a function that raises an error, is it meant to be used
in a for loop in a full study etc.
closed 2019-07-11 by agramfort