mne-tools/mne-python #6518: Incomplete PyVista 3d backend

GuillaumeFavelier, 2019-07-01:
As suggested in this [comment](https://github.com/mne-tools/mne-python/pull/6464#issuecomment-504440519), we keep track of the primitives not yet implemented in the PyVista 3d backend:

* `tube()`
* `text3d()`
* `scalarbar()`
* `_set_3d_view()`
* `_set_3d_title()`

GuillaumeFavelier, 2019-07-03:
Showing the advancement with the example `examples/connectivity/plot_sensor_connectivity.py` :

Mayavi | PyVista
----------|-----------
[image] | [image]

agramfort, 2019-07-03:
Next is color bar then ?

GuillaumeFavelier, 2019-07-03:
yes indeed!

GuillaumeFavelier, 2019-07-05:
In order to add `_set_3d_view()` and `_set_3d_title()` in the pyvista 3d backend, I have to use the `BackgroundPlotter` instead of the regular `Plotter` which depends on `PyQt5`. 
I need to thoroughly investigate the impact of this migration on the `_Brain` object, the interactivity inside a jupyter

agramfort, 2019-07-05:
let me know if you need to discuss

>

closed 2019-08-02 by GuillaumeFavelier