mne-tools/mne-python #6523: import dataset

Hamidbme, 2019-07-03:
I need to import a dataset for use in mne-python, but can not find a way to do so. I want to import this dataset in mne for preprocessing.

dataset: Jaeyoung Shin, Alexander von Lühmann, Benjamin Blankertz, Do-Won Kim, Han-Jeong Hwang and Klaus-Robert Müller, "Open Access Dataset for EEG+NIRS Single-Trial Classification," IEEE Trans. Neural Syst. Rehabil. Eng., online published. Download dataset here.

Could anyone help me in this issue please?
Thank for help.

cbrnr, 2019-07-03:
According to the description, the data is stored in .mat files. You can use `scipy.io.loadmat` to load it. This will give you a NumPy array, which you then need to convert to a `Raw` MNE object using https://mne-tools.github.io/stable/generated/mne.io.RawArray.html (see also [this example](https://mne-tools.github.io/stable/auto_examples/io/plot_objects_from_arrays.html)).

drammock, 2019-07-03:
duplicate of #6508. As I said in the follow-up email thread, please follow the guidance in the [getting help](https://mne-tools.github.io/stable/pre_install.html#getting-help) section of our website. Specifically: use [the MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) and/or [MNE gitter channel](https://gitter.im/mne-tools/mne-python) for questions about usage. The GitHub issues are for feature requests and bug reports.

closed 2019-07-03 by drammock