mne-tools/mne-python #6527: Warning in compute covariance

ktavabi, 2019-07-03:
`compute_covariance(epochs)` generates `RuntimeWarning`...

> Something went wrong in the data-driven estimation of the data rank as it exceeds the theoretical rank from the info (298 > 43). Consider setting rank to "auto" or setting it explicitly as an integer.

`compute_covariance(epochs, rank='full')` throws `ValueError` as rank must be 'full' or 'info'. What exactly is the `RuntimeWarning` referring to? `method` argument input or instance of epochs if someone could clarify I'm happy to open PR.

agramfort, 2019-07-03:
can you share the epochs data with a code to replicate the pb? thx

>

ktavabi, 2019-07-03:
[Here](https://www.dropbox.com/s/n1iz0dfmamu59uq/All_55-sss_bad_116a-epo.fif?dl=0) you go...
```
from mne import (read_epochs, compute_covariance)
from mne.epochs import combine_event_ids

eps = read_epochs('All_55-sss_bad_116a-epo.fif')
eps_copy = combine_event_ids(eps.copy(),
                             [k for k in eps.event_id.keys()],
                             {'All': 123})
signal_cov = compute_covariance(eps_copy)
```

agramfort, 2019-07-04:
I could replicate the error. We'll take a closer the look.
Use rank='info' as it will use the true rank derived from
maxfilter header. If you do

mne.viz.plot_cov(signal_cov, eps_copy.info)

you'll see that the rank of the data is closer than 34
than 301.