mne-tools/mne-python #6528: raw.plot(block=True) freezes the jupyter kernel forever

wmvanvliet, 2019-07-04:
#### Describe the bug

When plotting continuous "raw" data and specify `block=True`, everything works well until you try to close the figure. Control is never handed back to the command line. This does not occur when using command line ipython, but does occur when using Spyder or a Jupyter notebook.

#### Steps and/or code to reproduce

In Spyder or a notebook:

```Python
import mne
data_path = mne.datasets.sample.data_path()
raw = mne.io.read_raw_fif(data_path +'/MEG/sample/sample_audvis_raw.fif')
raw.plot(block=True)
```

Close the plotting window. The Jupyer kernel now hangs.

#### Expected results
After closing the window, the Jupyter kernel should become responsive again.

#### Actual results
Jupyter kernel never becomes responsive.

#### Additional information

```
Platform:      Linux-4.4.0-148-generic-x86_64-with-debian-stretch-sid
Python:        3.6.8 |Anaconda, Inc.| (default, Dec 30 2018, 01:22:34)  [GCC 7.3.0]
Executable:    /work/modules/Ubuntu/14.04/amd64/common/anaconda3/latest/envs/neuroimaging/bin/python
CPU:           x86_64: 8 cores
Memory:        15.6 GB

mne:           0.18.1 (also tested this on current master)
numpy:         1.16.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.1.0 {backend=Qt5Agg}

sklearn:       0.21.2
nibabel:       2.4.1
mayavi:        4.6.2 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.24.2
dipy:          Not found
```

agramfort, 2019-07-04:
can you replicate with a pure matplotlib code?

Kriattiffer, 2019-07-08:
You may need to switch backedns with
`%matplotlib qt`
or 
`%matplotlib tk`
when you need interactive plot

wmvanvliet, 2019-07-09:
Yup, this is a jupyter/matplotlib issue. This also fails with the same behaviour on my end:

```python
from matplotlib import pyplot as plt
plt.plot([1, 2, 3, 4])
plt.show(block=True)
```

closed 2019-07-09 by wmvanvliet

agramfort, 2019-11-19:
@wmvanvliet @SophieHerbst just reported the same idea. Did you end up solving this pb?

wmvanvliet, 2019-11-19:
Unfortunately, no.

mvonflotow, 2021-08-03:
I was able to fix this for myself by disabling "Active support" for Matplotlib (under tools -> preferences -> IPython console -> Graphics), then manually setting the Matplotlib backend via:

import matplotlib
matplotlib.use('Qt5Agg')
# or: %matplotlib qt5, but Spyder complained about this version.

Let me know if this works for you @wmvanvliet

CarinaFo, 2021-08-31:
Did anyone solve this issue for jupyter notebook? I am using matplotlib.qt and I get the mentioned kernel freeze.

aroyphillips, 2022-10-12:
I would like to reopen this issue as well for jupyter notebook. Is there a solution?

cbrnr, 2022-10-12:
I cannot reproduce the issue, the minimal example does not block execution in both a regular Jupyter notebook and inside VS Code. Maybe you do not have the latest versions of the involved packages (i.e. Matplotlib, Jupyter)?

wmvanvliet, 2022-10-12:
This still breaks for me in Spyder, but not in plain `ipython` or in a notebook:

```python
from matplotlib import pyplot as plt
plt.plot([1, 2, 3, 4])
plt.show(block=True)
```

However, @mvonflotow's suggestion of disabling Spyder's active support did the trick. I'm guessing there's some weird interaction between the Qt thread spawned by the `QtConsole` component that Spyder uses and the `block=True` behavior. I'll file the issue there.