mne-tools/mne-python #6531: Marking bad channels and bad ICs is inconsistent

cbrnr, 2019-07-05:
When plotting a `raw` object, I can mark bad channels by either clicking on a channel trace or on a channel label. When I do the same thing with ICs, the behavior is inconsistent in two ways:

1. Bad (excluded) ICs turn red instead of gray
2. I can only click on an IC trace, but not on an IC label (which tries to open a topomap, but fails if no montage is loaded).

I'd like to make this consistent, but we need a way to graphically trigger a topomap from a selected component. Maybe Shift-Click on the label would work for that?

agramfort, 2019-07-05:
sounds good to me !

>

mmagnuski, 2019-07-05:
I actually like the current behavior: clicking the label to get topomap is quite convenient. So if we were to make it consistent I would prefer that clicking the channel label in raw plotter would give the topographical position of the channel (or maybe even channel properties? :) ). The downside is that it is usually more difficult to select a signal trace in the raw plotter because if you are not accurate enough you end up marking the time sample with green line (at least that used to be some time ago). Because of this I much prefer to mark bad channels by clicking the channel labels. This is not the case in ICA signal plotter because there is no green line there (at least I don't think I ever experienced its presence).
The different colors could also be justified to some extent: gray when marking channel as bad could be visually read as "ignore", while the red for ICA component means "remove" (from the data). Gray is less visually intrusive so when one has a very bad channel jumping up and down across the whole signal plotter then marking that channel with gray looks ok while red could make the plott less readable. The ICA plotter on the other hand does not have this problem because the component signals are scaled independently.

agramfort, 2019-07-05:
I am not so sure now :)

>

drammock, 2019-07-05:
FWIW, on an `epochs.plot()` you left-click the channel name to mark a bad channel, and right-click a channel name to get a popup of `epochs.plot_image()` for that channel.  So one way of increasing consistency a little would be to make the ICA plot more like epochs plot (left-click channel name -> mark bad, right-click channel name -> show topoplot).  Adding a pop-up behavior to `raw.plot()` channel names (as @mmagnuski suggests) could then also be done on right-click.

mmagnuski, 2019-07-05:
@drammock - good idea. I actually wasn't aware of the right-click epochs image behavior.

cbrnr, 2019-07-08:
If this is how `epochs.plot` already works, then I think it's a good idea to streamline the other plots. I'm OK with leaving the color as is (red), although we could make this configurable at some point.

To summarize, here are the proposed changes:
- Left-clicking on an IC trace or an IC label should toggle the exclude status of that IC.
- Right-clicking on an IC label should open a topomap (or IC properties window, whatever is the current behavior).

Optionally, in a separate PR we could also add a right-click on channel label behavior for `raw.plot` - the question is what should be triggered. Any ideas?

cbrnr, 2019-07-08:
PS: Although from a usability point of view, I'd prefer Shift + left click instead of right click, but this means we'd have to change the behavior of `epochs.plot` too.

agramfort, 2019-07-08:
> To summarize, here are the proposed changes:

   - Left-clicking on an IC trace or an IC label should toggle the exclude
   status of that IC.
   - Right-clicking on an IC label should open a topomap (or IC properties
   window, whatever is the current behavior).

+1 for me

cbrnr, 2019-07-09:
Another inconsistency:
Right-clicking in the signals area in `epochs.plot` moves the green line (which is actually yellow)
Shift-left-clicking in the signals area (not on a trace) in `raw.plot` moves the green line (which is actually green)

larsoner, 2019-09-11:
@cbrnr will you be able to get to this in the next week? If so it can make it into 0.19, otherwise we'll push to 0.20

drammock, 2019-09-11:
if @cbrnr doesn't have time in the nex week I could try to do this at the sprint.

larsoner, 2020-03-04:
@cbrnr @drammock I'm going to bump the milestone on this to 0.21 since it does not seem like a blocker, but feel free to implement in the next couple of weeks to get it into 0.20

drammock, 2020-11-17:
closed by #8381

closed 2020-11-17 by drammock