mne-tools/mne-python #6544: BrainVision event parser doesn't handle my data correctly

hoechenberger, 2019-07-09:
The event parser for BrainVision files generates seemingly arbitrary event codes for my data. This is a problem that was first mentioned in #6267 and supposedly fixed. However, the fix introduced special handling depending on the detected event type:

https://github.com/mne-tools/mne-python/blob/70789c466133eaba410e1fc254c4b88723e30b77/mne/io/brainvision/brainvision.py#L825-L829

In the `vmrk` files generated in our lab using PyCorder, _all_ stimulus and response events are simply labeled as `Event` (i.e., not `Stimulus` and `Response`, respectively). Therefore, they get assigned IDs `10001+`, and the IDs we assigned to them during recording are basically ignored, yielding the issue(s) brought up in #6267.

When I change the line
```python
_BV_EVENT_IO_OFFSETS = {'Stimulus/S': 0, 'Response/R': 1000, 'Optic/O': 2000}
```
to include `Event` markers:
```python
_BV_EVENT_IO_OFFSETS = {'Stimulus/S': 0, 'Event/': 0, 'Response/R': 1000, 'Optic/O': 2000}
```
all works as expected and I can do
```python
data = mne.io.read_raw_brainvision('/Users/hoechenberger/Google Drive/TaLa/EEG_Analysis_outdated/1_DataEEG/Sub_721_071217/TaLa_721_01.vhdr')
events, _ = mne.events_from_annotations(data)
event_id = {'fixation': 100,
            'stim_a':   110,
            'stim_b':   120,
            'stim_c':   160,
            'response': 230}
epochs = mne.Epochs(data, events=events, event_id=event_id)
```
as would be expected.

larsoner, 2019-07-09:
Seems reasonable to me. Do you happen to know if `Event` is basically the same thing as `Stimulus`, or are they different? Maybe just from different versions of PyCorder?

hoechenberger, 2019-07-09:
> Seems reasonable to me. Do you happen to know if `Event` is basically the same thing as `Stimulus`, or are they different? Maybe just from different versions of PyCorder?

I'm not sure, maybe it's just a configuration setting. The data were recorded two years ago and I currently don't have a working EEG setup to run further tests. But I _could_ try to get a copy of PyCorder working again.

I would, intuitively, assume that `Event` is just a generic term for pretty much _all_ kinds of stimulus and response events.

hoechenberger, 2019-07-09:
@larsoner Maybe your question was actually whether the offset for `Event`s should be zero or something else? The answer is, I believe it should always be zero in that case. :)

agramfort, 2019-07-09:
I am fine with the suggested 1 line change.

you'll need to update a test though

thx

>

hoechenberger, 2019-07-09:
> you'll need to update a test though

I'd add a new sample file to `mne/io/brainvision/tests/data` and add a very small test that _just_ tests whether `Event` markers are being read correctly, nothing else, would that suffice?

larsoner, 2019-07-09:
Maybe just modify an existing `vhdr` file to have `Event/` rather than `Stimulus/`, that way you don't even need to add a new file (or even a new test, since it should show up the same way)

hoechenberger, 2019-07-09:
> Maybe just modify an existing `vhdr` file to have `Event/` rather than `Stimulus/`

You mean, replacing some existing `Stimulus/S` markers with `Event/`?

agramfort, 2019-07-09:
>
> You mean, replacing some existing Stimulus/S markers with Event/?
>
yes

closed 2019-07-10 by larsoner