mne-tools/mne-python #6547: Get more outputs from rap_music

makkostya, 2019-07-10:
`rap_music` function currently returns list of dipole objects and residuals. Dipole object only contains the information about dipole position and orientation in euclidean coords.  It's not however  possible to know to which source from the source space this dipole corresponds, its hemisphere and what is its leadfield. 

It would be useful to have a possibility to get the source index and corresponding topography as an output of `rap_music`:

```
dipoles, idx, hemi, lf = rap_music(evoked, fwd, noise_cov, n_dipoles=3, 
                                   return_source_idx=True,
                                   return_leadfields=True)
```

By the way, the source index is printed in logs in current implementation:
`
source 1 found: p = 14659`

makkostya, 2019-07-10:
Another possibility is to return stc object instead of dipoles if needed

larsoner, 2019-07-10:
+1 for return_type='dipoles' (default) or 'stc'

makkostya, 2019-07-10:
Ok, I'll try to do it.

agramfort, 2019-07-10:
great thanks !

>