mne-tools/mne-python #6552: ENH: Warn if distances are bad

larsoner, 2019-07-10:
After excluding outliers, we should warn somewhere if people's coregistration is bad. Candidate automatic computation places:

1. `plot_alignment`
2. `make_forward_solution`
3. `mne.Report`

Bad criteria could be > 5 mm (?) for either:

1. median distance
2. "trimmed mean" distance, where instead of trimming upper and lower, we just trim upper (25%? 50%?)

This can be a thin wrapper around the function implemented in #6512.

cc @jhouck in case you have ideas

jhouck, 2019-08-23:
There are probably ways that a coregistration could be bad/questionable in some way that's not obvious from the median distance -- e.g. if one of the polhemus sensors shifted during subject preparation, or if too few extra points were collected, or maybe if the number of dropped outlier points exceeds some threshold.

If the extra points were sampled broadly enough, it may be possible to test whether the HPI points were close to the surface defined by the head points, but that would depend pretty heavily on the person doing the subject preparation.  Also that's going to be an ugly surface.

larsoner, 2019-08-24:
Agreed there are many possible failure modes. I think we should start with something simple like median distance, and iterate on adding additional checks when we find cases where it doesn't capture a given problematic case.

agramfort, 2019-08-24:
don't forget that we have now the dig_mri_distances function

>

jhouck, 2019-08-24:
That reminds me.  During coregistration, outlier points are often dropped to improve the fit.  Information about which points are dropped/kept is not retained.  Does this matter?  `dig_mri_distances` doesn't use all points, so maybe it's fine -- and at least in https://14892-1301584-gh.circle-artifacts.com/0/dev/auto_examples/forward/plot_auto_alignment.html the error is lower for `dig_mri_distances` than for `_get_point_distance`.  I should probably change that example to show the number of points in each, but after dropping 3 outlier points `_get_point_distance` reports on 140 points, and `dig_mri_distances` on 72.

agramfort, 2019-08-25:
no strong feeling here. Whatever you think is best

>