mne-tools/mne-python #6556: EEG header file

Mehrabkhazraei, 2019-07-11:
Hello every one,
I have a EEG data recorded by myself. now i checked mne can read header file of type vhdr(vmrk), but i have V-amp file with extension of ahdr(amrk). do anyone know how can i convert it or how can i read these type of files??
for your convenience you can check them by the link below.
https://drive.google.com/drive/folders/1eRnBBU6Ne0Ahl8I7z6k9tHCIl0AxQ0T1?usp=sharing

sappelhoff, 2019-07-11:
Hi @Mehrabkhazraei this is a good question for the Brain Products customer support. I have had a good experience with them in the past. Just write them an email to: support@brainproducts.com and explain your problem.

And don't forget to post the solution here once you have obtained it! :-)

Mehrabkhazraei, 2019-07-11:
Hi, @sappelhoff . what you mean by good experience?? did you find any solution for this? :-)

sappelhoff, 2019-07-11:
No but I have had questions similar to yours. These questions were either ...

- about Brain Products equipment
- Brain Products software
- ... and sometimes even unrelated to a Brain Products product

They usually answer within one to two days. Given your question, there is a high likelihood that they will solve your problem.

drammock, 2019-07-11:
I agree with @sappelhoff that emailing brain products is the right step to take first. But FYI, the google drive link you posted is for an empty folder, so we can't examine or test your files anyway.

sappelhoff, 2019-07-23:
Any news on this @Mehrabkhazraei ?

Mehrabkhazraei, 2019-07-24:
Hello Estefan,

 

As I understood there is no way to convert V-amp(yet). But you can process your data using matlab more easily. If you need any help don’t be hesitate to contact me.

sappelhoff, 2019-07-24:
Sorry to hear that @Mehrabkhazraei ... thanks for the update and let us know if you want to write a conversion tool yourself

closed 2019-07-24 by sappelhoff