mne-tools/mne-python #6562: RFC: suppress filter params output in some cases?

drammock, 2019-07-11:
The new output summarizing filter params (added in #6294) is really cool, but `create_ecg_epochs` and `create_eog_epochs` each involve two filtering steps, and the filter params are not fully under user control, so the output is maybe less necessary in those cases. I wonder if it makes sense to have a way to suppress the `filter_data` messages without fully disabling all `create_ecg_epochs` messages?  Ideas:

- if `verbose` is `None` in the call to `create_ecg_epochs`, automatically pass `verbose=False` to the `filter_data` call.  User can still see all the messages by passing `verbose=True`, and will see no messages (from either `create_ecg_epochs` or `filter_data`) by passing `verbose=False`.
- a new boolean param in `create_ecg_epochs` that gets passed to `filter_data`'s `verbose` param, called `show_filter_params` or `filter_verbose` or something similar.

larsoner, 2019-07-11:
Or just always pass verbose=False in the find_ecg_events call. Users probably don't need to know the characteristics used to filter just for the purposes of finding the events. And if they want this information they can make a separate call to find_ecg_events

larsoner, 2019-07-11:
(the parameters that are used to filter the raw data before epoching the data should arguably always be with the rest of the logging messages, because they reflect the filtering characteristics of the Epochs you get back)

drammock, 2019-07-11:
Not sure I fully understand your second comment @larsoner...  I thought you were saying

> maybe we should always suppress the filter messages when `filter_data` is called from within `create_ecg_epochs`

but then you said the parameters should arguably *always* be there?

larsoner, 2019-07-11:
When filtering the ecg signal no output, when filtering the raw data to epoch, yes output

drammock, 2019-07-11:
> When filtering the ecg signal no output, when filtering the raw data to epoch, yes output

OK, agreed.

larsoner, 2019-07-11:
So I think the only change you should need to make is passing verbose=False to the find_ecg_events call the occurs within create_ecg_epochs

drammock, 2019-07-12:
yup.  I'll wait half a day to let others weigh in before implementing.

agramfort, 2019-07-12:
Ok for me

PR #6565 (2019-07-12): suppress filter param messages when not user-specifiable
closes #6562 

I ended up taking a middle-road approach: hardcoding `verbose=False` where the user has no control over filter params, but leaving things as-is for filters where the params are passed through from the calling function. Now a call to `create_ecg_epochs` will spit out only one set of filter params instead of two (yes for the initial filtering step, but not for the call to `qrs_detector`). Passing `verbose=False` to `create_ecg_epochs` will of course suppress both filter param messages (as well as all the other ones like "rec [... cut]
closed 2019-07-12 by larsoner