mne-tools/mne-python #6566: VIZ, BUG: test_plot_alignment timeout

GuillaumeFavelier, 2019-07-12:
As noticed by @drammock , the `test_plot_alignment()` in `mne/tests/test_3d.py` seems to [timout after 120 seconds](https://travis-ci.org/mne-tools/mne-python/jobs/558019119#L3570) (limit fixed currently) on Mac. Without confirmation if it's spurious or not, I [experienced it myself](https://github.com/mne-tools/mne-python/pull/6551) and it would be annoying to break Travis systematically. So for now, we see several ways to fix this:
* increase timeout
* refactor the test either in less calls with the same coverage
* split in multiple tests
* use the `slow_test` decorator

What do you think @massich , @larsoner ?

larsoner, 2019-07-12:
Slow test is my vote. 2 minutes is a long time

agramfort, 2019-07-13:
agreed. +1 for slow test

PR #6567 (2019-07-14): FIX: Mark test_plot_alignment as slowtest
Fixes #6566 by marking `test_plot_alignment()` as `slowtest`.
closed 2019-07-14 by agramfort