mne-tools/mne-python #6579: epochs.pllot_drop_log

drammock, 2019-07-16:
`epochs.plot_drop_log()` returns a figure saying "no drops" in cases where percentage of drops is lower than `threshold` parameter. There are two problems with this:

1. it is inaccurate (there were drops)
2. it is wrong behavior (according to the docs, `threshold` > percent dropped is supposed to *suppress* the figure, but it still gets generated / returned)

`epochs.plot_drop_log()`

[image]

`epochs.plot_drop_log(threshold=5)`

[image]

drammock, 2019-07-16:
while I'm at it, I'll center the bars over the ticks, and make the bars slightly darker so they're more visible on low-brightness laptop screens (unless someone objects)

agramfort, 2019-07-17:
+100

larsoner, 2019-07-17:
Sounds good

commit a7b39388 (2019-07-18): closes #6579 (#6582)
closed 2019-07-18 by agramfort