mne-tools/mne-python #6581: BUG: events_form_annotation does not work with brainvision files

kcsaba12, 2019-07-17:
def find_events(filename):
    # raw = mne.io.read_raw_edf(filename, preload=True)
    raw = mne.io.read_raw_brainvision(filename, preload=True)
    raw.rename_channels(lambda x: x.strip('.'))
    events, _ = mne.events_from_annotations(raw, event_id=dict(T0=1, T1=2, T2=3))
    print(events)

With the given function I would like to read the events of a brainvision file.
I got the following error message:

Traceback (most recent call last):
  File "D:/Users/Csabi/dev/Cybathlon/sandbox.py", line 273, in <module>
    find_events(files[1])
  File "D:/Users/Csabi/dev/Cybathlon/sandbox.py", line 242, in find_events
    events, _ = mne.events_from_annotations(raw, event_id=dict(T0=1, T1=2, T2=3))
  File "<C:\Programs\Miniconda3\envs\bci\lib\site-packages\mne\externals\decorator.py:decorator-gen-56>", line 2, in events_from_annotations
  File "C:\Programs\Miniconda3\envs\bci\lib\site-packages\mne\utils\_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "C:\Programs\Miniconda3\envs\bci\lib\site-packages\mne\annotations.py", line 912, in events_from_annotations
    annotations.description, event_id=event_id, regexp=regexp)
  File "C:\Programs\Miniconda3\envs\bci\lib\site-packages\mne\annotations.py", line 820, in _select_annotations_based_on_description
    raise ValueError('Could not find any of the events you specified.')
ValueError: Could not find any of the events you specified.

When I remove the event_id parameter, events, _ = mne.events_from_annotations(raw), everything goes fine:
C:\Programs\Miniconda3\envs\bci\python.exe D:/Users/Csabi/dev/Cybathlon/sandbox.py
Extracting parameters from D:/BCI_stuff/databases/Cybathlon pilot/pilot1/rec01.vhdr...
Setting channel info structure...
Reading 0 ... 1232999  =      0.000 ...  2465.998 secs...
Used Annotations descriptions: ['New Segment/', 'Response/R  1', 'Stimulus/S  1', 'Stimulus/S  2', 'Stimulus/S  3', 'Stimulus/S  4', 'Stimulus/S  5', 'Stimulus/S  6', 'Stimulus/S  7', 'Stimulus/S  8', 'Stimulus/S  9', 'Stimulus/S 10', 'Stimulus/S 11', 'Stimulus/S 12', 'Stimulus/S 14', 'Stimulus/S 15']
[[      0       0   99999]
 [  34602       0      14]
 [  40122       0    1001]
 ...
 [1222720       0       5]
 [1224720       0      12]
 [1224821       0      15]]

Moreover the given function works fine even with edf files.

larsoner, 2019-07-17:
> When I remove the event_id parameter, events, _ = mne.events_from_annotations(raw), everything goes fine:

It looks like your `event_id` above:
```
event_id=dict(T0=1, T1=2, T2=3)
```
Is not correct for this data, but should look more like:
```
event_id={'Stimulus/S 1': 1, 'Stimulus/S 2': 2, 'Stimulus/S 3': 3}
```

kcsaba12, 2019-07-17:
Thanks!! now I get it. 
I think the updated web description confused me:
https://mne-tools.github.io/stable/auto_examples/decoding/plot_decoding_csp_eeg.html#sphx-glr-auto-examples-decoding-plot-decoding-csp-eeg-py

I tired to compare 
events = find_events(raw, shortest_event=0, stim_channel='STI 014')    - 0.17.2 description
with
events, _ = events_from_annotations(raw, event_id=dict(T1=2, T2=3))   - 0.18.1 description

I though that the event_id parameter is the same as the parameter in the
epochs = Epochs(raw, events, event_id, tmin, tmax, proj=True, picks=picks,
                baseline=None, preload=True)
therefore I could rename the triggers. But now I see that these parameters are not the same.

Maybe rename one of them would be beneficial.

Thanks for the help!

closed 2019-07-17 by larsoner

agramfort, 2019-07-17:
it's basically the same meaning for event_id except that we use
dict(hands=2, feet=3) instead of dict(T1=2, T2=3) to explicit what T1 and
T2 correspond to

>