mne-tools/mne-python #6583: ValueError: Channel XXX has no associated original unit when channel names trucated

bdyetton, 2019-07-18:
### Describe the bug
When channels names in an EDF are long, they are truncated. 
When there are two channels that are truncated, then they get appended with 0, 1, 2 etc
Because of this appended number, the channel is no longer in the orig_units_trunc, and the Value error is thrown.

#### Steps and/or code to reproduce
```Python
mne.io.read_raw_edf('any edf file with long channel names which will be trucated to create non unique channel names.edf')
```

..\..\..\Anaconda3\lib\site-packages\mne\io\base.py:407: ValueError
```Python
            # Each channel in the data must have a corresponding channel in
            # the original units.
            ch_correspond = [ch in orig_units_trunc for ch in ch_names] #issue is here
            if not all(ch_correspond):
                ch_without_orig_unit = ch_names[ch_correspond.index(False)]
                raise ValueError('Channel {} has no associated original '
                                'unit.'.format(ch_without_orig_unit)) 
throws: ValueError: Channel EMG Ant Tibia-0 has no associated original unit.
```

agramfort, 2019-07-18:
hum... can you share a file so we can replicate?

>

bdyetton, 2019-07-18:
Yup, here is a shortened version:

[MASS_SS1_subjectid1_001.zip](https://github.com/mne-tools/mne-python/files/3406993/MASS_SS1_subjectid1_001.zip)

agramfort, 2019-07-18:
see https://github.com/mne-tools/mne-python/pull/6586

bdyetton, 2019-07-18:
Works beautifully, cheers :)

PR #6586 (2019-07-19): FIX : EDF reading with long names that are the same for the first 15c
closes #6583 (I think)

cc @bdyetton 

can you check if it works for you?
closed 2019-07-18 by bdyetton