mne-tools/mne-python #6584: EEG Units not converted to Volts

bdyetton, 2019-07-18:
#### Describe the bug
Hi MNE team,
I'm still quite sure what is going on, but i believe that the EEG channels in the attached EDF are not being correctly translated to volts.
I've opened the file in a number of programs, and the values are different in each.

#### Steps and/or code to reproduce
the file:
[unit_testing.zip](https://github.com/mne-tools/mne-python/files/3407335/unit_testing.zip)

```Python
import mne
raw = mne.io.read_raw_edf('unit_testing.edf')
some_edf_channels = ['C3','C4']
chan_data = raw.get_data(some_edf_channels)
print(np.sum(np.abs(chan_data))/chan_data.size) #this prints 0.035V, which is too high, it should be 0.000035V
```

agramfort, 2019-07-18:
file has a problem:

ipdb> orig_units

{'C3': 'mV', 'C4': 'mV', 'M1': 'mV', 'M2': 'mV', 'LOC': 'mV', 'ROC': 'mV',
'ECG2': 'uV', 'ECG1': 'uV', 'EMG 1': 'uV', 'EMG 2': 'uV', 'EMG 3': 'uV', 'L
Leg': 'mV', 'R Leg': 'mV', 'AIRFLOW': 'mV', 'THOR EFFORT': 'mV', 'ABDO
EFFORT': 'mV', 'SNORE': 'mV', 'SUM': 'mV', 'POSITION': '', 'OX STATUS': '',
'PULSE': 'BPM', 'SpO2': '%', 'NASAL PRES': 'V', 'PlethWV': 'V', 'Light':
'', 'HRate': 'bpm'}

who wrote this file?

you can still hack the raw._data manually after using preload=True

>

bdyetton, 2019-07-18:
Hi,
I got it from the NSRR: https://sleepdata.org/, specifically from the Cleaveland Family Study.
Sorry, but what's the issue in the ```orig_unit```s here? the mixed voltage types? 

Thanks, I'll hack if necessary, but I'd rather fix the file if possible? 
I guess i could hack it in MNE and then resave as EDF.

agramfort, 2019-07-19:
as you see it says from edf header than units are mV and not uV for the EEG
channels

so I suspect header is wrong

>

bdyetton, 2019-07-21:
Hmm, ok but the abs values read by other programs are correct:
```Python
    import pyedflib
    import numpy as np
    f = pyedflib.EdfReader('C:/Users/bdyet/Desktop/unit_testing_eeg_only.edf')
    n = f.signals_in_file
    pyedf_sigs = np.zeros((n, f.getNSamples()[0]))
    for i in np.arange(n):
        pyedf_sigs[i, :] = f.readSignal(i)
    print('pyedf av in mV:', np.sum(np.abs(pyedf_sigs)) / pyedf_sigs.size) 

    from wonambi import Dataset
    wnbi = Dataset('C:/Users/bdyet/Desktop/unit_testing_eeg_only.edf')
    wnbi_sigs = wnbi.read_data().data[0]
    print('Wonambi av in mV:', np.sum(np.abs(wnbi_sigs))/wnbi_sigs.size)

    import mne
    raw = mne.io.read_raw_edf('C:/Users/bdyet/Desktop/unit_testing_eeg_only.edf') #prints ~35.0
    mne_sigs = raw.get_data()
    print('MNE av in mV:', np.sum(np.abs(mne_sigs))/(1e3*mne_sigs.size))
```
which prints:
```
pyedf av in mV: 0.035007964494163435
Wonambi av in mV: 0.03500796449416343
MNE av in V: 0.03500796449416343
```
Both wonambi and pyedflib do no convertion (i.e. preseving units as mV).
How can these be the same values when MNE does a conversion to volts? It seems to me that mne is not doing the conversion.

agramfort, 2019-07-22:
see https://github.com/mne-tools/mne-python/pull/6589

indeed you were right. It was broken.

bdyetton, 2019-07-22:
Awesome, working now, thanks :)
```
pyedf av in mV: 0.035007964494163435
Wonambi av in mV: 0.03500796449416343
MNE av in V: 3.500796449416343e-05
```

closed 2019-07-22 by bdyetton

agramfort, 2019-07-23:
thanks for the clear bug report !

PR #6589 (2019-07-22): FIX: unit mV in EDF reader
closes #6584 

cc @bdyetton