mne-tools/mne-python #6587: Can't import EDF file: AssertionError

levchenkoegor, 2019-07-19:
Hi,

I am trying to open raw EEG file in EDF format. 

`import mne`
`data = mne.io.read_raw_edf(r"C:\Users\CDM\Documents\elevchenko\Sync_EEG_NT\example.edf")`

But the next error occurs:

`AssertionError:` 

I am attaching this file.
[example.zip](https://github.com/mne-tools/mne-python/files/3411400/example.zip)

I am using '0.18.2' version of MNE.

I don't understand where the problem is, but I guess something with encoding.

I will be very thankful for any comments.

agramfort, 2019-07-22:
this file is likely to be corrupted. Neither MNE nor pyedflib
https://pyedflib.readthedocs.io/en/latest/ can read it.

>

levchenkoegor, 2019-07-22:
That's strange because I have 6 different files and I can open them via BrainStorm in Matlab. I'll ask my colleagues for the other file, but I am not sure that all of the files can be corrupted.

agramfort, 2019-07-23:
I also get an error using EEGLAB reader based on biosig. Dimension
mismatch + a bunch of warnings.

>

levchenkoegor, 2019-07-23:
Yes, but without importing EDF anotations EEGLAB is loading it properly.

I asked colleagues to upload files again, but the same error is. They mentioned that they used this EEG amplifier:
https://www.advancedbrainmonitoring.com/xseries/x24/

agramfort, 2019-07-24:
what do the annotations contain?

levchenkoegor, 2019-07-30:
Sorry for the delay. I tried to understand does annotations have something specific, but I think it leads the basic protocol for EDF files:
"The annotation channel contains an ASCII encoded representation of the time in seconds that certain events happened"

agramfort, 2019-07-30:
sorry but i have no idea how to help. See if you can read it with any open source package and look what we do differently.

cbrnr, 2019-08-05:
I checked this file with [EDFbrowser](https://www.teuniz.net/edfbrowser/), [pyedflib](https://pyedflib.readthedocs.io/en/latest/), and [BioSig](http://biosig.sourceforge.net/), and it really looks like this is not a valid EDF file. The header seems to contain invalid information, such as the 30th character of the PID field is not a valid ASCII character and the number of bytes in the header do not match the expected signals. Indeed, BioSig can load the file without annotations, but it shows a weird sample frequency of 256.0164 Hz - I don't know how much this result can be trusted, plus you don't get the annotations either. If you can live with that, the best way to proceed is probably to load the files with BioSig (or EEGLAB) and store the result in a valid file format that MNE can read (e.g. EEGLAB .set).

levchenkoegor, 2019-09-23:
Just a little updated - the problem was that colleagues send me these files not raw, but after manipulations in their software. This software is making format of the file incompatible with other toolboxes as I understood.

I think it is can be closed.

closed 2019-09-23 by levchenkoegor

larsoner, 2019-09-23:
Great, thanks for following up @levchenkoegor