mne-tools/mne-python #6592: BUG?  ICA.plot_sources has side effects

drammock, 2019-07-23:
running `ica.plot_sources(raw, exclude=[2, 3])` will append indices 2 and 3 to `ica.exclude`.  This seems to be a result of the "click the trace to mark for exclusion" functionality.  It is quite surprising for this to happen when passing the `exclude` parameter and simply viewing the plot non-interactively.  This never comes up with `raw.plot()` because `raw.plot()` doesn't have a `bads` parameter to the plot method.

Not sure what the best solution is. Some options:

1. get rid of `exclude` parameter from `ica.plot_sources` to make it consistent with `raw.plot` (i.e., `exclude` is read from the instance attribute). Retain click-to-mark-bad behavior.

2. keep current method signature, but change code so that any indices passed in to `exclude` are ignored when updating `ica.exclude` after closing the plot. This seems bad because clicked-on components (red) will be added to `exclude`, but passed-to-method components (also red) will not be.

3. keep current method signature, but change code so that any indices passed in to `exclude` are ignored when updating `ica.exclude` after closing the plot *unless* at least one component was clicked on, in which case treat clicked components and passed-to-method components equivalently. (i.e., add an internal flag for whether a component was toggled, and only update `ica.exclude` if that flag is True). This seems bad too, because sometimes red components are added to `exclude`, sometimes they're not.

4. keep everything as-is, but just make the side-effect clearer in the docs. This seems bad because side-effects are bad, period. The ICA instance shouldn't be modified by a `plot_*` method call if the user doesn't interact with the plot in any way besides looking at it with their eyes.

drammock, 2019-07-23:
```ipython
In [35]: ica.exclude
Out[35]: []
In [36]: eog_indices
Out[36]: [0]
In [37]: ica.plot_sources(raw, exclude=eog_indices)  # don't click anything, just look & close the plot
Out[37]: <Figure size 1920x1040 with 4 Axes>
In [38]: ica.exclude
Out[38]: [0]
```

larsoner, 2019-07-23:
I'd prefer option 1. Users can easily append to exclude then plot

agramfort, 2019-07-23:
+1 to simply remove the exclude param in ica.plot_sources

maybe @cbrnr has an opinion?

PR #6595 (2019-07-25): MRG: Fix side effects in ica.plot_sources
closes #6592
closed 2019-07-25 by larsoner